TheThingsNetwork/lorawan-stack #2913: Pass configuration options to frontend

htdvisser, 2020-07-15:
#### Summary

We need to find a way to pass backend configuration to the frontend.

#### Why do we need this?

- So that the frontend can give users more information about the actions they perform.
- So that the frontend can disable functionality that isn't enabled on the backend.

#### What is already there? What do you see now?

- We have some configuration in the Identity Server related to user signup (password requirements, invite only, admin approval required).
- We have configuration in the Identity Server related to disabling picture uploads

#### What is missing? What do you want to see?

- The Account frontend needs those configuration options so that it can show those password requirements, require an invitation token, or inform the user that they need to wait for admin approval.
- The Account and Console frontends need to know if picture uploads are enabled

#### How do you propose to implement this?

Option A: Pass such configuration in the HTML template. This works for the Account frontend (because it runs on the Identity Server, but not for the Console, because the Identity Server runs on a different server.

Option B: Add a method to the `ttn.lorawan.v3.Configuration` service to get config options (this has my preference).

#### How do you propose to test this?

Can be unit-tested

#### Can you do this yourself and submit a Pull Request?

Yes, if I have time, but if anyone is interested in working on this, please comment!

KrishnaIyer, 2020-07-15:
+1 for Option B.

bafonins, 2020-07-16:
Are we talking about these https://github.com/TheThingsNetwork/lorawan-stack/blob/develop/pkg/webui/template.go#L93-L101 ?

I guess option A is the easiest. However, option B seems to be more appropriate and reliable. It is also quite helpful for mocking stack configuration for e2e tests.

kschiffer, 2020-07-20:
Agreed. Option B.

htdvisser, 2020-08-10:
With the API done, I'm handing this over to @kschiffer and/or @bafonins.

closed 2020-11-03 by bafonins