TheThingsNetwork/lorawan-stack #2923: Constant password change on 'Create new user' afterlogin on v3.4.6 until page refresh.

thevirtualdj, 2020-07-16:
Constant password change on 'Create new user' afterlogin on v3.4.6 until page refresh. 

Implemented the stack on public server and on Create new user everything is fine but after we log in user has to change pass or logout. 
On every option after change or logout, when we log in again it says change password again. 
[image]

KrishnaIyer, 2020-07-16:
Please upgrade your version to the latest. We've `3.8.7` released yesterday. We don't provide fixes to versions older than the previous minor (currently lesser than `3.7.x`)

KrishnaIyer, 2020-07-16:
Oops. Wrong button

thevirtualdj, 2020-07-16:
Sorry i made a typo 3.4.6 - 3.8.6. Will try and close the issue. 
Ty.

thevirtualdj, 2020-07-16:
[image]
Still having the same problem? 
Did I missconfigured something or?

bafonins, 2020-07-16:
Can confirm that the issue is present in the latest `master`. Basically, the problem here is when creating a new account first we lose the `next` url in the request, which results in always logging in the oauth app regardless if the user visited `host/console` or `host/oauth`. @kschiffer is this something to be implemented in https://github.com/TheThingsNetwork/lorawan-stack/issues/1422 ?

kschiffer, 2020-07-20:
Indeed, this is essentially a duplicate of TheThingsIndustries/lorawan-stack#1968 (this should be an open source issue...)  and is something that really really needs to be fixed ASAP.

@thevirtualdj we are aware of this issue and looking into possible fixes.

kschiffer, 2020-07-20:
> [**@kschiffer**](https://github.com/kschiffer) is this something to be implemented in [#1422](https://github.com/TheThingsNetwork/lorawan-stack/issues/1422) ?

As far as retaining the `next` query parameter goes, yes, I will make sure to add this.

thevirtualdj, 2020-08-12:
Still present in 3.9.0

johanstokking, 2020-08-24:
@thevirtualdj we're working on it, please bear with us.

closed 2021-03-01 by johanstokking