TheThingsNetwork/lorawan-stack #2925: SMTP Email client notification settings thru 465 ?

thevirtualdj, 2020-07-16:
Can I and how to set up smtp thru SSL conection port 465?
Here is my smtp config: 
                                       --is.email.provider="smtp"
                                       --is.email.sender-address="noreply@test.com"
                                       --is.email.sender-name="test"
                                       --is.email.sendgrid.api-key=""
                                       --is.email.sendgrid.sandbox="false"
                                       --is.email.smtp.address="mail.your-server.de"
                                       --is.email.smtp.connections="0"
                                       --is.email.smtp.password="mysecretpayy" (here is my pass ofc in real env)
                                       --is.email.smtp.username="noreply@test.com" (here is my username ofc in real env - email is real username on our server)

htdvisser, 2020-07-29:
If you set `is.email.smtp.address` to `"mail.your-server.de:465"`, SSL is automatically enabled.

htdvisser, 2020-07-29:
Adding documentation label because we could document this better in the configuration reference.

thevirtualdj, 2020-08-12:
Thank you that did solved the problem.

closed 2020-08-12 by thevirtualdj