TheThingsNetwork/lorawan-stack #2926: Application packages console support

adriansmares, 2020-07-16:
#### Summary

We need to support creating/editing/deleting application (default) package associations from the Console.

#### Why do we need this?

Such that people can setup the packages without using the CLI.

#### What is already there? What do you see now?

1. Application packages documentation + CLI support details:
https://github.com/TheThingsNetwork/lorawan-stack/blob/79d20616887b274b582d826431733f26fd38c41e/doc/content/integrations/application-packages/_index.md
https://github.com/TheThingsNetwork/lorawan-stack/blob/79d20616887b274b582d826431733f26fd38c41e/doc/content/integrations/application-packages/lora-cloud-device-and-application-services.md
2. Application package (default) associations API: 
https://github.com/TheThingsNetwork/lorawan-stack/blob/008bc54e7a9f5169f0ca235f4dbb77deaa92a76e/api/applicationserver_packages.proto#L113-L178
3. Application package (default) association CLI support:
https://github.com/TheThingsNetwork/lorawan-stack/blob/008bc54e7a9f5169f0ca235f4dbb77deaa92a76e/cmd/ttn-lw-cli/commands/applications_packages.go#L382-L427

#### What is missing? What do you want to see?

Console support for setting up the associations (CRUD). Realistically we should start with default associations (which are basically application wide associations between an application and a specific package).  Device specific associations (which are overrides) are less important at the moment

#### Environment

`3.9.0`

#### How do you propose to implement this?

They should be situated under the Pub/Sub integrations, named `Application Packages`. The initial screen should follow what we already have for normal webhooks - i.e. a list of associations. When I click one I can edit the package and the options of the package. Options (the `data` field of an association) are specific to an association, but the showing the actual fields instead of waiting JSON input from the user would be better. 

As is the case with our API, you can expect that if a package named `lora-cloud-device-management-v1` has a field `token`, the field will always be there .

How can the Console obtain an application package list ? At this time it should be hardcoded, since you have different fields for each package, and since we want to show them to the user instead of just a JSON text input, we have to add custom support anyway.

#### How do you propose to test this?

Testing if CRUD works fine is enough.

#### Can you do this yourself and submit a Pull Request?

No. @kschiffer and @bafonins please reassign this based on availability, and remove the discussion label once all of the details are clear.
cc @johanstokking

johanstokking, 2020-07-16:
@adriansmares how about creating a specific LoRa Cloud DAS integration page, that has the package name, port and input for token sort of hard coded? I think the flexibility of the packages in the backend is good, but in the frontend we should make this really as easy as possible. This way, we can also validate the token and we don't have to enumerate the package list.

adriansmares, 2020-07-16:
> @adriansmares how about creating a specific LoRa Cloud DAS integration page, that has the package name, port and input for token sort of hard coded? I think the flexibility of the packages in the [...]

Port is still user input, but otherwise this is also fine 👍 

I think this is also to keep in mind for other integrations - maybe tables are practical but the UX is a lot better with 1 click setup.

johanstokking, 2020-07-17:
> I think this is also to keep in mind for other integrations - maybe tables are practical but the UX is a lot better with 1 click setup.

Yes, see https://github.com/TheThingsIndustries/lorawan-stack/pull/2241#issuecomment-657323916

johanstokking, 2020-07-17:
Setting high priority on this one due to partner request.

@kschiffer my idea would be to show "LoRa Cloud" as a first-class integration, not in pub/sub (it isn't pub/sub anyway), so on the same level as MQTT, webhooks and pub/sub. It would work with a fixed FPort.

We may later add an overview of package assignments per FPort, but I'd like to keep this issue small and scoped to configuring LoRa Cloud in a way that is future proof.

johanstokking, 2020-07-21:
I simplified the original comment here.

@kschiffer for the UX, I think it would be nice to have in the section:

- a logo
- a short description
- a link to documentation
- a link to Semtech's resources
- input fields for token

johanstokking, 2020-07-30:
What is the status here?

laurensslats, 2020-08-03:
@kschiffer do you already have an initial design implementation? I can send you the required info (logo, description etc) if needed.

@adriansmares do you have some pointers so we can create a documentation on https://thethingsstack.io/. I can help out with writing docs and testing the integration.

kschiffer, 2020-08-03:
> do you already have an initial design implementation? I can send you the required info (logo, description etc) if needed.

No initial impementation yet, but I have all questions clarified now so that I should be able to implement this swiftly. Getting logo, description and resources from you would be nice.

laurensslats, 2020-08-03:
a logo
They don't have an officiel logo. You can use this (this is the only visual I got), or leave it out completely.
[image]

a short description:

**LoRa Cloud™ Device & Application Services**

With the LoRa Cloud Device & Application Services protocol, you can manage common device functionality at the application layer for LoRaWAN®-enabled devices. This protocol consists of a set of messages that are exchanged on a predefined device management LoRaWAN port. The purpose of these messages is three-fold:
1. Periodically communicate info messages
2. Trigger client-initiated management commands
3. Run advanced, application-layer protocols which solve common LoRaWAN use cases

a link to documentation:
https://thethingsstack.io/latest/integrations/application-packages/lora-cloud-device-management/

a link to Semtech's resource:
https://www.loracloud.com/documentation/device_management

@kschiffer

PR #3047 (2020-08-11): Add LoRa Cloud integration to the Console
#### Summary
Closes #2926 

[image]

#### Changes
- Add application package service to JS SDK
- Add LoRa Cloud form container
- Add LoRa Cloud integration view
- Add store logic for application packages
- Add feature toggle for application packages
- Add a couple of new hooks and utility components to cater for views written as functional components
- Small related styling improvements

#### Testing

- Manual testing

#### Notes for Reviewers
I've taken the opportunity to try out some new approaches regarding
- Writing some hook-versions of our HOCs
-  [... cut]
closed 2020-08-11 by kschiffer