TheThingsNetwork/lorawan-stack #2927: Enums.yml missing comments

benolayinka, 2020-07-16:
#### Summary

Enums with a description in https://github.com/TheThingsNetwork/lorawan-stack/blob/master/api/api.md do not have any description in https://github.com/TheThingsNetwork/lorawan-stack/blob/master/doc/data/api/ttn.lorawan.v3/enums.yml and so the descriptions are not available in the docs.

#### What is missing? What do you want to see?

Can we generate enums.yml and the other api yaml documents the same way we generate api.md? With comments? @htdvisser 

#### Can you do this yourself and submit a Pull Request?

Maybe

htdvisser, 2020-07-16:
Those files are generated, but that needs to be done manually until @rvolosatovs has time for https://github.com/TheThingsIndustries/docker-protobuf/issues/33.

This is what I use myself:

```bash
brew install protobuf
go install htdvisser.dev/exp/protoget
go install htdvisser.dev/exp/protoc-gen-hugodata
protoget -dst third_party google/api/annotations.proto google/api/http.proto github.com/gogo/protobuf/gogoproto/gogo.proto github.com/envoyproxy/protoc-gen-validate/validate/validate.proto
protoc -I $(pwd)/third_party -I $(dirname $(pwd)) --hugodata_out=output_path=doc/data:doc/data $(pwd)/api/*.proto
```

---

Which enums exactly don't have the comment you'd expect?

benolayinka, 2020-07-16:
> Those files are generated, but that needs to be done manually until @rvolosatovs has time for [TheThingsIndustries/docker-protobuf#33](https://github.com/TheThingsIndustries/docker-protobuf/issues/33).
> 
> This is what I use myself:
> 
> ```shell
> brew install protobuf
> go install htdvisser.dev/exp/protoget
> go install htdvisser.dev/exp/protoc-gen-hugodata
> protoget -dst third_party google/api/annotations.proto google/api/http.proto github.com/gogo/protobuf/gogoproto/gogo.proto github.com/envoyproxy/protoc-gen-validate/validate/validate.proto
> protoc -I $(pwd)/third_party -I $(dirname $(pwd)) --hugodata_out=output_path=doc/data:doc/data $(pwd)/api/*.proto
> ```
> 
> Which enums exactly don't have the comment you'd expect?

`AggregatedDutyCycle` is missing the comment

htdvisser, 2020-07-16:
Aha I see. Fixed that in the generator: https://github.com/htdvisser/exp/commit/f9ec2ecce670c47fca9dd11d7949d6eef61f34e2

closed 2020-07-20 by benolayinka