TheThingsNetwork/lorawan-stack #2935: Include completion files in Brew builds

htdvisser, 2020-07-17:
#### Summary

It would be nice to add completion files to Brew builds.

#### Why do we need this?

So that completion is pre-installed instead of asking users to generate it.

#### What is already there? What do you see now?

#2870 added support for writing completion files

#### What is missing? What do you want to see?

Completion included in our Brew packages 

#### How do you propose to implement this?

- Update build process to write completion to `completion/ttn-lw-{cli,stack}/{bash,zsh}`
- Update Goreleaser to include completion in archives
- Update Brew install part to (I think) something like
    ```ruby
    output = Utils.safe_popen_read(bin/"completion", "ttn-lw-cli", "bash")
    (bash_completion/"ttn-lw-cli").write output
    output = Utils.safe_popen_read(bin/"completion", "ttn-lw-cli", "zsh")
    (zsh_completion/"_ttn-lw-cli").write output
    ```

#### How do you propose to test this?

I think you can test installing Brew files locally

neoaggelos, 2020-07-17:
There is an issue for this already https://github.com/TheThingsNetwork/lorawan-stack/issues/2907

And some preliminary work in https://github.com/TheThingsNetwork/lorawan-stack/tree/feature/snap-and-brew-autocomplete

htdvisser, 2020-07-17:
Aha okay. This was still on my action list from the sync call, didn't notice the other issue.

closed 2020-07-17 by htdvisser