TheThingsNetwork/lorawan-stack #2938: Device import by manifest

johanstokking, 2020-07-17:
#### Summary

Device import format

#### Why do we need this?

For device makers to share a manifest with their customers, which contains all information for The Things Stack to provision the end devices. The manifest acts as a proof of ownership.

The manifest may contain root keys, but these must be encrypted using asymmetric cryptography, with the public key of the targeting Join Server. This would typically be the Global Join Server.

#### What is already there? What do you see now?

Device Template Converter infrastructure to convert files to end device templates. There is support in the Console and CLI to upload files. The Console executes the templates and can create the end devices immediately, the CLI does that too and supports intermediate steps.

#### What is missing? What do you want to see?

1. [ ] Design manifest file format
2. [ ] Add RPC for DTC (JS cluster) to share the public key
3. [ ] Add CLI commands to templates to create the manifest file
4. [ ] Add converter to DTC

#### How do you propose to implement this?

1. JWE per end device entry. The contents would be an `EndDeviceTemplate` in our JSONPB format
2. Add rpc to `EndDeviceTemplateConverter`
3. Take public key, take templates from `stdin` and write file with JWEs. We might want to make it easy and work with an lookup-CSV to get root key(s) per DevEUI
4. Implement JWE decryption and return templates

#### How do you propose to test this?

Local

#### Can you do this yourself and submit a Pull Request?

Yes, can do, but feel free to pick this up

neoaggelos, 2021-08-18:
Is this related to/duplicate of https://github.com/TheThingsNetwork/lorawan-stack/issues/3769?

johanstokking, 2021-08-19:
Not really; that's more about proof of ownership in files. I'm closing this because we can work with our JSON format, which is well documented.

closed 2021-08-19 by johanstokking