TheThingsNetwork/lorawan-stack #2953: Validation of pub/sub form doesn't work

johanstokking, 2020-07-22:
#### Summary

Validation of pub/sub form doesn't work

#### Steps to Reproduce

1. Configure new pub/sub
2. Select MQTT
3. Enter invalid client ID, i.e. with spaces

#### What do you see now?

[image]

#### What do you want to see instead?

Validation errors.

They are defined, but they don't fire in the Console. For example:

https://github.com/TheThingsNetwork/lorawan-stack/blob/db2df0298fabc4e06c09e2321c29fb865ab7d1ae/pkg/webui/console/components/pubsub-form/validation-schema.js#L76-L80

Note that none of these validators are effective.

#### Environment

3.8.7, also current `master`

#### How do you propose to implement this?

Not sure

#### How do you propose to test this?

Test locally, enter invalid fields, observe validation errors in Console

#### Can you do this yourself and submit a Pull Request?

Can review

bafonins, 2020-07-22:
Cannot reproduce. Validation works fine on the latest master.

We previously had issues with the pub-sub form, but I fixed this in https://github.com/TheThingsNetwork/lorawan-stack/pull/2860

bafonins, 2020-07-22:
However one thing is broken. The error message is not really helpful:
[image]

The following diff should fix the validation error message:
```diff
@ pkg/webui/console/components/pubsub-form/validation-schema.js
	export default Yup.object().shape({
         .matches(mqttUrlRegexp, sharedMessages.validateUrl)
         .required(sharedMessages.validateRequired),
       client_id: Yup.string()
-        .matches(noSpacesRegexp, sharedMessages.validateNoSpaces)
+        .matches(noSpacesRegexp, Yup.passValues(sharedMessages.validateNoSpaces))
         .min(2, Yup.passValues(sharedMessages.validateTooShort))
         .max(23, Yup.passValues(sharedMessages.validateTooLong))
         .required(sharedMessages.validateRequired),
```
[image]

johanstokking, 2020-07-22:
Should we always `passValues()` or when do we do this?

bafonins, 2020-07-23:
> Should we always passValues() or when do we do this?

Usually, we use `Yup.passvalues` to include values into the validation message. For example, when we enforce that a certain field must be at least 2 characters long we use the `.min(2)` rule and define a single validation message `{field} must have at least {min} characters` such that validation value is placed instead of `{min}`. 

The problem with `string().matches()` is that its signature is different from other methods `matches(regex, options)` where `options` is an object `{message: string, excludeEmptyString: bool}` while other methods simply accept a string. The issue can be solved with the following approach as well without the use of `passValues`:

```diff
@ pkg/webui/console/components/pubsub-form/validation-schema.js
	export default Yup.object().shape({
         .matches(mqttUrlRegexp, sharedMessages.validateUrl)
         .required(sharedMessages.validateRequired),
       client_id: Yup.string()
-        .matches(noSpacesRegexp, sharedMessages.validateNoSpaces)
+        .matches(noSpacesRegexp, {message: sharedMessages.validateNoSpaces})
         .min(2, Yup.passValues(sharedMessages.validateTooShort))
         .max(23, Yup.passValues(sharedMessages.validateTooLong))
         .required(sharedMessages.validateRequired),
```
`passValues` implicitly sets the validation method into the `message` field of the `options` object.

bafonins, 2020-07-23:
@johanstokking Can we close this issue?

johanstokking, 2020-07-23:
Yes, thanks for the explanation

closed 2020-07-23 by johanstokking