TheThingsNetwork/lorawan-stack #2962: Document setting up TTS on localhost for evaluation/testing purposes

neoaggelos, 2020-07-22:
#### Summary

Document how users can start the stack on localhost for testing purposes and for getting a quick taste of The Things Stack.

#### Why do we need this ?

Currently, the Getting Started guide has instructions for setting up TTS on a public instance under `thethings.example.com`. There are also a few kinda sparse notes for local deployments.

Users may want to quickly get a TTS instance up and running on `localhost` quickly for evaluation purposes, and the documentation is currently lacking in that regard. This can lead to confusion for new users.

#### What is already there? What do you see now?

- The Getting Started guide, for setting up TTS on a public server, and a few notes for local deployments.
- Instructions for setting up a local development environment in `DEVELOPMENT.md`.

#### What is missing? What do you want to see?

Instructions for setting up a TTS installation on `localhost`. This does not have to be too novel or extraordinary, rather a list of simple steps to very quickly get TTS up and running on localhost, without having to download and setup a full development environment.

Maybe also have some kind of "quick start" commands, in order to demonstrate creating an application, a gateway, an end device and simulating a few LoRaWAN packets.

Again, the purpose here would be to make it super simple for new users to setup and evaluate TTS.

#### How do you propose to document this?

Add a new section named something like "Running on localhost". Avoid duplicating existing docs, add links to existing documentation where needed.

#### Can you do this yourself and submit a Pull Request?

Yes, also cc @benolayinka

KrishnaIyer, 2020-07-23:
just replacing `thethings.example.com` to `localhost:<port>` is sufficient. Why do we need a separate guide for that?

benolayinka, 2020-07-27:
> just replacing `thethings.example.com` to `localhost:<port>` is sufficient, right? Why do we need a separate guide for that?

The [getting started guide](https://thethingsstack.io/v3.8.7/getting-started/installation/) is a lot more work than the couple steps in DEVELOPMENT.md. It's also [easy to mess up](https://thethingsstack.io/v3.8.7/getting-started/installation/configuration/) the yaml config. And it's not clear what certificates you need if you're running on localhost.

@neoaggelos maybe it's enough to link to DEVELOPMENT.md for users who want to run a dev stack really quickly?

KrishnaIyer, 2020-07-27:
> And it's not clear what certificates you need if you're running on localhost.

You don't use TLS on localhost. It's actually meaningless to do that even though we can generate certs. TLS is for when something is exposed on the public internet.

benolayinka, 2020-07-27:
> > And it's not clear what certificates you need if you're running on localhost.
> 
> You don't use TLS on localhost. It's actually meaningless to do that even though we can generate certs. TLS is for when something is exposed on the public internet.

Exactly, but the first time I followed the getting started guide, I tried to use ACME to generate certs because there's nothing about localhost

neoaggelos, 2020-07-27:
> > And it's not clear what certificates you need if you're running on localhost.
> 
> You don't use TLS on localhost. It's actually meaningless to do that even though we can generate certs. TLS is for when something is exposed on the public internet.

I believe that the source of confusion here is that even though on localhost you will not be using TLS for interacting with the stack, the stack itself still requires a certificate to be setup, and will refuse to start without one.

This might not be clear enough to a newcomer that wants to try out the stack. Also, they might not be familiar enough with generating a self-signed certificate.

> @neoaggelos maybe it's enough to link to DEVELOPMENT.md for users who want to run a dev stack really quickly?

That is also a solution, but having to setup a development environment may be too much work for this, or not even an option for some people.

johanstokking, 2020-08-24:
For evaluating the stack, users will be able to use TTN.

Otherwise, people know how things work, can use the existing docs for a local deployment.

closed 2020-08-24 by johanstokking