TheThingsNetwork/lorawan-stack #2964: Pub/Sub integration providers are not retrieved in the Console

adriansmares, 2020-07-23:
#### Summary

Pub/Sub integration information page (so the Pub/Sub form when you first enter a Pub/Sub for editing) does not retrieve the provider fields (MQTT/NATS).

#### Steps to Reproduce

1. Create a Pub/Sub integration
2. Fill in the provider information (use MQTT for example).
3. Leave the page, go to the Pub/Sub list and select your integration.
4. You won't see the provider information.

#### What do you see now?

The provider information is not visible.

#### What do you want to see instead?

The provider information should be visible.

#### Environment

Cloud Hosted, `3.8.7`

#### How do you propose to implement this?

I've ran some tests and if I use the `provider` field mask as opposed to the two `provider.mqtt` and `provider.nats`, I do get the information. This means:

`https://<edited>.industries/api/v3/as/pubsub/<edited>/brbrbr?field_mask=base_topic,format,provider.nats,provider.mqtt,downlink_ack,downlink_failed,downlink_nack,downlink_push,downlink_queued,downlink_replace,downlink_sent,join_accept,location_solved,uplink_message` - does not retrieve the MQTT provider.

`https://wilhelmsen.eu1.cloud.thethings.industries/api/v3/as/pubsub/e2e-demo-001/brbrbr?field_mask=base_topic,format,provider,downlink_ack,downlink_failed,downlink_nack,downlink_push,downlink_queued,downlink_replace,downlink_sent,join_accept,location_solved,uplink_message` - does retrieve the MQTT provider.

#### How do you propose to test this?

Check if the console correctly displays the provider information after leaving and entering the page again.

#### Can you do this yourself and submit a Pull Request?

I can review this.

johanstokking, 2020-07-28:
I cannot reproduce.

> 4\. You won't see the provider information.

Please explain.

[image]

This contains provider information right?

adriansmares, 2020-07-29:
For reference, this issue was not reproducible in `v3.9`, so we consider it to be fixed for now.

closed 2020-07-29 by adriansmares