TheThingsNetwork/lorawan-stack #2965: CLI Auth Error - Following Change to TLS Redirection in Config

ZeroSum24, 2020-07-23:
#### Summary

Using ttn-lw-cli v3.8.7 I'm unable to run console commands against a self-managed TTN v3.8.7 deployment. This was working previously and has since stopped following the addition of the http.redirect-to-tls flag in the deployment configuration.

The console is still accessible and working as expected. I am not sure if this is a bug or a system configuration issue. Either way your help would be much appreciated!

#### Steps to Reproduce

1. Install dockerised TTN deployment on a remote server with a Let's Encrypt certificate
2. Follow the getting started guide on setting up the deployment, adding in the http.redirect-to-tls flag, set to true
3. Install the ttn-lw-cli toolset on another machine, and point the CLI to use the server domain
4. Try to run a command using the setup ttn-lw-cli toolset

**ttn-lw-stack.yml - Server:**

# Identity Server configuration
 is:
  # Email configuration for "REMOVED"
   email:
     sender-name: 'The Things Stack'
     sender-address: 'noreply@REMOVED'
     network:
       name: 'TTN-stack'
       console-url: 'https://REMOVED/console'
       identity-server-url: 'https://REMOVED/oauth'

 # Web UI configuration for "REMOVED":
   oauth:
      ui:
        canonical-url: 'https://REMOVED/oauth'
        is:
          base-url: 'https://REMOVED/api/v3'

 # HTTP server configuration
 http:
   cookie:
     block-key: '681d743cce27874de7728b184635592256057cafabaa5ffbc7030bd8a3c9c886'                # generate 32 bytes (openssl rand -hex 32)
     hash-key: '7963c721ed07de190da62539655f685c073baae2b207bd54074d8ce2f63d290365bda0c01365669929262d31e94b3c5d4ff7df8155c811588142acaa7af64417'                 # generate 64 bytes (penssl rand -hex 64)
   metrics:
     password: 'metrics'               # choose a password
   pprof:
     password: 'pprof'                 # choose a password
   redirect-to-tls: 'true'

# Let's encrypt for "REMOVED"
 tls:
   source: 'acme'
   acme:
     dir: '/var/lib/acme'
     email: 'REMOVED'
     hosts: ['REMOVED']
     default-host: 'REMOVED'

 #If Gateway Server enabled, defaults for "REMOVED":
 gs:
   mqtt:
     public-address: 'REMOVED:1882'
     public-tls-address: 'REMOVED:8882'
   mqtt-v2:
     public-address: 'REMOVED:1881'
     public-tls-address: 'REMOVED:8881'

 #If Gateway Configuration Server enabled, defaults for "REMOVED":
 gcs:
   basic-station:
     default:
       lns-uri: 'wss://REMOVED:8887'
   the-things-gateway:
     default:
       mqtt-server: 'mqtts://REMOVED:8881'
[... 355 lines cut ...]
      },
      "code": 2
    }
  ]
}

#### What do you want to see instead?

Ideally, I would like to get authorised and be able to login via the console.

#### Environment

...

#### How do you propose to implement this?

...

#### How do you propose to test this?

...

#### Can you do this yourself and submit a Pull Request?

Nope, but I very much appreciated your help!

neoaggelos, 2020-07-23:
Hi @ZeroSum24 , thanks for reporting this issue. Indeed, I was able to reproduce the problem. This has been around since `3.8.0` apparently.

ZeroSum24, 2020-07-23:
@neoaggelos thanks very much for your help and the quick response! I've applied that workaround on our end which has resolved the issue for the moment.

neoaggelos, 2020-08-25:
Should be fixed with #3120

closed 2020-08-25 by neoaggelos