TheThingsNetwork/lorawan-stack #2969: Ubuntu based Docker images

adriansmares, 2020-07-23:
#### Summary

We should provide Docker images deriving from `ubuntu` as opposed to only `alpine`.

#### Why do we need this?

In order to have this variant available for testing / prototyping. 

I've read a couple of reports (https://github.com/grafana/grafana/issues/20153, for example)  that `alpine` based images have some weird DNS resolver issues, especially when combined with Go (which sometimes uses the system resolver and sometimes uses the fully-Go-native resolver). Reports claim that no such issues exist under `ubuntu` based images, so I'd definitely wish to have these available to debug errors (such as the failing link errors, which always seem to be linked to lingering DNS records for some reason).

#### What is already there? What do you see now?

Our image derives `alpine`, most likely since it is a very small base image in comparison to `ubuntu`.

#### What is missing? What do you want to see?

An `ubuntu` derived Dockerfile specification, and adding the said Dockerfile to Goreleaser.

#### Environment

`3.8.7`, but not important.

#### How do you propose to implement this?

`apk` will have to be switched with `apt` (and make sure not to leave the `sources.list` as part of the image) and most likely the syntax for `addgroup` and `adduser` is different under `ubuntu` (iirc).

#### How do you propose to test this?

Run a `ubuntu` based image in a staging environment.

#### Can you do this yourself and submit a Pull Request?

Yes. cc @htdvisser

htdvisser, 2020-07-23:
I don't think this is the way to go. Having Ubuntu or even Alpine around a static binary is unnecessary. To be honest, I would rather switch to images `FROM scratch`.

If we want predictable DNS behavior we can build our binaries with the `netgo` tag, so that we always use Go's native DNS implementation.

htdvisser, 2020-07-23:
Found my WIP on the `FROM scratch` idea:

```Dockerfile
FROM alpine as builder

RUN apk --update add ca-certificates
COPY ttn-lw-stack /bin/ttn-lw-stack
RUN ln -s /bin/ttn-lw-stack /bin/stack
RUN chmod 755 /bin/ttn-lw-stack
COPY ttn-lw-cli /bin/ttn-lw-cli
RUN ln -s /bin/ttn-lw-cli /bin/cli
RUN chmod 755 /bin/ttn-lw-cli
COPY public /srv/ttn-lorawan/public
RUN mkdir /srv/ttn-lorawan/public/blob

FROM scratch

COPY --from=builder /etc/ssl/certs/ca-certificates.crt /etc/ssl/certs/ca-certificates.crt
COPY --from=builder /bin/ttn-lw-* /bin/
COPY --from=builder /srv/ttn-lorawan/public/blob /srv/ttn-lorawan/public/blob
VOLUME ["/srv/ttn-lorawan/public/blob"]
EXPOSE 1700/udp 1881 8881 1882 8882 1883 8883 1884 8884 1885 8885 1887 8887
```

adriansmares, 2020-07-23:
> I don't think this is the way to go. Having Ubuntu or even Alpine around a static binary is unnecessary. To be honest, I would rather switch to images `FROM scratch`.
>
> If we want predictable DNS [...]

I'm ok with this variant as well. Shall we split then this into two issues ?
- Move to `FROM scratch` 
- Use the `netgo` tag as part of `goreleaser`

htdvisser, 2020-07-23:
I don't think we'll make the community very happy if we go `FROM scratch`. Doing that makes debugging a lot harder. 

Additional disadvantage is that we'd have to reimplement the health check (because scratch doesn't have `curl`).

closed 2021-03-23 by adriansmares