TheThingsNetwork/lorawan-stack #2973: Contact Info in the Web UI

htdvisser, 2020-07-24:
#### Summary

We should add functionality to the Console (and/or account app) to list and edit contact info, and to (re-)request validation of that contact info.

#### What is already there? What do you see now?

The top-level entities in The Things Stack can have contact info (typically email addresses, in the future maybe also phone/twitter/...) attached to them (although in practice this is only used by users right now). Contact info can be retrieved/updated by selecting the `contact_info` field of the entity.

Contact info looks like this:

```js
[{
  "contact_method": "CONTACT_METHOD_EMAIL",
  "value": "hylke@thethingsindustries.com",
  "validated_at": "2020-02-02T20:20:20Z" // if the contact info was validated.
}]
```

Contact info can be validated by sending an email with a validation token/link which the user then sends back to The Things Stack as proof that they have access to that email address. This email is requested with the `/ttn.lorawan.v3.ContactInfoRegistry/RequestValidation` call (`POST /api/v3/contact_info/validation`). The method takes `EntityIdentifiers` as the request message, and will request validation of all (not already validated) contact info. The returned `ContactInfoValidation` message contains an `id` field, which is the "reference". In the `contact_info` field you can see which email addresses were sent a validation token/link.

The CLI can list contact info with `ttn-lw-cli users get <user-id> --contact-info` or `ttn-lw-cli users contact-info list htdvisser`. The CLI can request validations with `ttn-lw-cli users contact-info request-validation <user-id>`. The same for `application`s, `client`s, `gateway`s and `organization`s.

#### What is missing? What do you want to see?

This functionality is missing from the console

#### How do you propose to implement this?

@kschiffer for UX

htdvisser, 2022-07-28:
In the meantime we've started replacing ContactInfo. See https://github.com/TheThingsNetwork/lorawan-stack/issues/4930 for details.

KrishnaIyer, 2024-10-22:
We support this in the console

closed 2024-10-22 by KrishnaIyer