TheThingsNetwork/lorawan-stack #2974: Add histogram metrics to Network Server

htdvisser, 2020-07-24:
#### Summary

It would be nice if we could add two histogram metrics to the Network Server:

- For DevAddr matches per uplink
- For the number of MIC checks per uplink

#### Why do we need this?

- DevAddr matches: So that we can monitor how well DevAddrs are spread, and see if we need to assign more blocks to a deployment.
- MIC checks: So that we can monitor how optimal our pre-sorting is.

#### How do you propose to implement this?

Prometheus already has support

#### How do you propose to test this?

🐒

rvolosatovs, 2020-07-25:
MIC check metrics are already part of #2837

nejraselimovic, 2021-06-07:
@htdvisser this seems useful for monitoring. Can you be the one to do this? I don't see it as a priority, but seems like it would be nice to have, so I'll schedule for Q3.

closed 2022-01-21 by adriansmares