TheThingsNetwork/lorawan-stack #2975: Replace otto with goja

htdvisser, 2020-07-24:
#### Summary

We should replace the unmaintained [otto](https://github.com/robertkrimen/otto) library with [goja](https://github.com/dop251/goja).

Refs #2670 and #2966

#### Why do we need this?

Because we should avoid unmaintained libraries, especially ones that we use to run arbitrary code.

htdvisser, 2020-07-24:
And while at it, it would be nice to add a Prometheus histogram to get some insight in how long those scripts take

johanstokking, 2020-07-27:
Let's do this in one go with rm ns-000013.json as.json

johanstokking, 2020-08-20:
I've spent a half day on this, and the conclusion so far is:

### https://github.com/dop251/goja

- maintained and they merge community PRs
- no Go module support (or any dep management whatsoever)
- no protection against stack overflows when exporting nested objects with circular reference (I have a fix for this)
- memory limit

### https://github.com/robertkrimen/otto 

- unmaintained for a long time
- new problems with Go 1.15
- relies on a fork to protect against stack overflows when exporting nested objects with circular references
- supports memory allocation limit

---

So, this neither are ideal. I would still opt to switch to `goja` because it is maintained and faster than `otto`.

I'm afraid we need to pick either to keep the builds simple.

PR #3152 (2020-08-27): Refactor AS message processors

#### Summary

Closes #2670
Closes #2975
Groundwork for https://github.com/TheThingsNetwork/lorawan-stack/issues/263

#### Changes

- Replace `otto` by `goja`
- Introduce input/output objects that give more flexibility, with backwards compatibility
- Decode downlink messages
- Retire device repository; it wasn't used and not documented anyway
- Change message processor API for external runners; it wasn't used and doesn't fit with new structure

#### Testing

Unit testing

##### Regressions

There should be no functional regressions.

A known performance regressio [... cut]
closed 2020-08-27 by johanstokking