TheThingsNetwork/lorawan-stack #2980: Show all event info in expanded event details

kschiffer, 2020-07-27:
#### Summary
The expanded details section of the console currently only shows the `data` prop of the event. However, there is other relevant data in the root of the event object that should be shown too.

#### Why do we need this?
The user should be able to inspect all available data of an event if he/she wishes to

#### What is already there? What do you see now?
Only the `data` prop of the event object being shown

#### What is missing? What do you want to see?
The whole event object being shown instead.

@bafonins @johanstokking has there been any discussion that led to this decision that I might have missed?

#### Environment
v3.8.x

#### How do you propose to implement this?
Just change the details component to show `event` instead of `event.data`

#### How do you propose to test this?
Trivial.

#### Can you do this yourself and submit a Pull Request?
Yes.

johanstokking, 2020-07-27:
> @bafonins @johanstokking has there been any discussion that led to this decision that I might have missed?

Not that I'm aware of. More information is always better. What do you have in mind to show?

kschiffer, 2020-07-28:
We can just show the whole event object instead of just `event.data`. Then we can also make all regular events expandable instead of just the ones that have a `data` prop.

PR #3229 (2020-09-22): Event fixes and improvements (Console)
#### Summary
This PR introduces fixes and improvements to event components and  views. It solves performance issues (#2887), [layout issues](https://github.com/TheThingsIndustries/lorawan-stack/issues/2260#issuecomment-682060901)), moves to show the complete event data on expand (#2980) and introduces "synthetic" status events (see #2884).

Closes #2887
Closes #2980
References #2260
References #2884

<details><summary>Screenshot</summary>

[image]
</details>

#### Changes
- Refactor event global and console components into a single event component
- Deter [... cut]
closed 2020-09-22 by kschiffer