TheThingsNetwork/lorawan-stack #3002: Use "Import end devices" for migrating devices to TTES

laurensslats, 2020-07-29:
#### Summary
More customers are migration their devices to TTES, from TTI V2 or a 3rd party NS.

Migrating devices can now only by using the [CLI](https://enterprise.thethingsstack.io/v3.8.7/getting-started/migrating-from-v2/). Not everyone is a big fan of using the the CLI, so we need to provide an option to add devices using the console.

#### Why do we need this?
Make it easy to migrate from V2 or any 3rd part LoRaWAN NS to TTES.

#### What is already there? What do you see now?
Import end devices feature is already present.

[image]

#### What is missing? What do you want to see?
- Create/define an end device template which people can select

Template should look something like this:
```
{
  "ids": {
    "device_id": "device_id",
    "application_ids": {
      "application_id": "application_id"
    },
    "dev_eui": "0000000000000000",
    "join_eui": "0000000000000000"
  },
  "name": "name_of_device",
  "lorawan_version":"1.0.2",
  "lorawan_phy_version":"1.0.2-b",
  "frequency_plan_id":"EU_863_870",
  "supports_join":true,
  "root_keys":{
      "app_key":{
          "key":"00000000000000000000000000000000"
        }
      },
      "mac_settings":{
          "rx1_delay":{
              "value":"RX_DELAY_1"
              },
            "supports_32_bit_f_cnt":true
        },
        "session":{
            "dev_addr":"00000000",
            "keys":{
                "app_s_key":{
                    "key":"00000000000000000000000000000000"
                },
                "f_nwk_s_int_key":{
                    "key":"00000000000000000000000000000000"
                }
            },
        "last_f_cnt_up":0,
        "last_n_f_cnt_down":0
    }
}
```

#### How do you propose to implement this?
- [ ] Get feedback from tech team
- [ ] Define end device template
- [ ] Implement end device template
- [ ] Write documentation

#### How do you propose to test this?
1. Test importing a .json file with a list of devices locally
2. Test importing a .json file with a list of devices on staging1

#### Can you do this yourself and submit a Pull Request?
- Need general feedback from @johanstokking or @htdvisser 
- Need input from @neoaggelos for the end device template
- Console implementation by @bafonins or @kschiffer 

cc @ajovertoom @rish-c

johanstokking, 2020-07-29:
I think this is a duplicate of https://github.com/TheThingsNetwork/lorawan-stack/issues/2938. Can you check? Or should we clarify that issue better?

laurensslats, 2020-07-30:
That issue might need clarification. For this issues, encrypted manifest files are not required I believe. I want to reduce the scope of this issue to 1 specified template, which aligns with the TTN V2 export .json file.

@neoaggelos are you able to take up this issue?

neoaggelos, 2020-07-30:
So, the issue here is that the device fields are stored separately (in AS/NS/IS and possibly JS). This requires some work in the Console. We would have to either replicate most of the work in the CLI for this (which handles splitting the device and storing the appropriate fields where they belong), or create a new endpoint (in IS?) that handles the device creation in AS/NS/JS, and have the console use that (using the `network_server_address`, `application_server_address` and `join_server_address` fields, along with proper documentation, should be enough to handle any use-case, like devices on the Global Join Server).

It most certainly should be do-able, but we need to be smart about this instead of patching together a quick-and-dirty solution. Adding `discussion` label.

bafonins, 2020-07-30:
> So, the issue here is that the device fields are stored separately (in AS/NS/IS and possibly JS). This requires some work in the Console. We would have to either replicate most of the work in the CLI for this (which handles splitting the device and storing the appropriate fields where they belong),

This is already implemented in the JS SDK. 

We can add a format for TTN V2 export .json file and reuse the current device import view.

johanstokking, 2020-07-30:
@neoaggelos I think what is needed is just a DTC formatter that returns the device as-is, but then as template, with relevant fields set (the known fields for V2).

johanstokking, 2020-07-30:
@laurensslats users still have to export devices from V2 with CLI.

johanstokking, 2020-08-10:
Conclusion here is that we need to have a DTC converter (that is always enabled) that basically takes the `EndDeviceTemplate` as-is, and returns the `EndDeviceTemplate`. That way, we can leverage the existing JS SDK and Console, functionality, and we can create additional model shaping in the future.

bafonins, 2020-08-10:
@johanstokking, @neoaggelos  Since there is no console/js sdk related work to be done could you assign the responsible person and adjust the labels?

johanstokking, 2020-08-10:
@bafonins I did that already; @neoaggelos will pick this up.

PR #3078 (2020-08-20): Implement device template converter for The Things Stack v3 JSON end device format

#### Summary

Closes #3002 

#### Changes

- Extract the `jsonDecoder` used in our CLI so that it can be reused.
- Implement `the-things-stack` device template converter.
- Enable converter by default.

#### Testing

Tested locally, successfully imported multiple devices from the Console "Import End Devices" page. I also want to add some simple unit tests for the converter.

##### Regressions

None

#### Notes for Reviewers

...

#### Checklist

- [X] Scope: The referenced issue is addressed, there are no unrel [... cut]
closed 2020-08-20 by neoaggelos