TheThingsNetwork/lorawan-stack #3042: Add Swagger UI

neoaggelos, 2020-08-06:
#### Summary

Add SwaggerUI for the v3 HTTP API.

#### Why do we need this?

Make it easier for users to explore and test the available HTTP APIs.

#### What is already there? What do you see now?

We generate `api.swagger.json`, which contains HTTP definitions for all services that are defined as protos (some GCS endpoints are not present there).

#### What is missing? What do you want to see?

Support for Swagger UI, so users can browse the API endpoints. A small "Browse the API" link on the Console could also be helpful.

#### Environment

3.9.0

#### How do you propose to implement this?

The first step would be to make the `api.swagger.json` available to users. It should be straightforward enough to add it as a static asset when building the frontend.

For SwaggerUI itself, There are a handful of options:
- Use the `swagger-ui-react` NPM package
- Add static `swagger-ui` assets, write some very little Javascript for handling API authentication.

We could then server swagger UI under `/api/v3/swagger`.

Not too well familiar with how we want to handle this on the Console, so cc @kschiffer @bafonins 

#### How do you propose to test this?

Verify that SwaggerUI is served properly, test some API calls using the "Try It out" buttons.

#### Can you do this yourself and submit a Pull Request?

Depends on how we decide to implement this.

johanstokking, 2020-08-09:
I don't think we should be self-hosting this from The Things Stack. If we host this, it would be in a managed distribution (like TTI Cloud Hosted), and then there are better ways with [a container](https://hub.docker.com/r/swaggerapi/swagger-ui/dockerfile) and configuring the proxy. So I don't think this really belongs here.

neoaggelos, 2020-10-19:
Closing this, let's reopen if there is interest.

closed 2020-10-19 by neoaggelos