TheThingsNetwork/lorawan-stack #3053: nil

veghead, 2020-08-08:
#### Summary

TTN stack 3.9.0 is currently crash-looping with a nil pointer dereference.

#### Steps to Reproduce

1. Start the Stack.

#### What do you see now?

```
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x38 pc=0x18d2aba]
goroutine 124 [running]:
go.thethings.network/lorawan-stack/v3/pkg/applicationserver.(*ApplicationServer).decryptDownlinkMessage(0xc000843a20, 0x2774260, 0xc000d68c90, 0xc0007baec0, 0x1b, 0xc000581e38, 0x8, 0x0, 0xc000581e40, 0xc000581e48, ...)
	/Users/veg/devel/lorawan-stack-backend/pkg/applicationserver/applicationserver.go:1029 +0x18a
go.thethings.network/lorawan-stack/v3/pkg/applicationserver.(*ApplicationServer).handleUp(0xc000843a20, 0x2774260, 0xc000d68c30, 0xc000227300, 0xc000646200, 0x5f2eda2b, 0xa92c0f0, 0x3b4e166d88bc6)
	/Users/veg/devel/lorawan-stack-backend/pkg/applicationserver/applicationserver.go:540 +0x637
go.thethings.network/lorawan-stack/v3/pkg/applicationserver.(*link).sendUp(0xc000646200, 0x2774260, 0xc000470b10, 0xc000227300, 0xc000825d28, 0xc000646200, 0x2792d40)
	/Users/veg/devel/lorawan-stack-backend/pkg/applicationserver/linking.go:348 +0x2c6
go.thethings.network/lorawan-stack/v3/pkg/applicationserver.(*ApplicationServer).link(0xc000843a20, 0x2774260, 0xc000470870, 0xc000945300, 0x8, 0x0, 0xc00094c4c0, 0x0, 0x0)
	/Users/veg/devel/lorawan-stack-backend/pkg/applicationserver/linking.go:258 +0x9bc
go.thethings.network/lorawan-stack/v3/pkg/applicationserver.(*ApplicationServer).startLinkTask.func1(0x2774260, 0xc000470870, 0x39b0920, 0x0)
	/Users/veg/devel/lorawan-stack-backend/pkg/applicationserver/linking.go:63 +0x229
go.thethings.network/lorawan-stack/v3/pkg/component.(*Component).StartTask.func1(0xc0004708a0, 0x2774260, 0xc000470870, 0x279e120, 0xc00094c0c0, 0x2, 0x38e0cd0, 0x3, 0x3, 0x3fb999999999999a)
	/Users/veg/devel/lorawan-stack-backend/pkg/component/tasks.go:82 +0x84
created by go.thethings.network/lorawan-stack/v3/pkg/component.(*Component).StartTask
	/Users/veg/devel/lorawan-stack-backend/pkg/component/tasks.go:77 +0x150
```
#### What do you want to see instead?

No crash

#### Environment

Version 3.9.0 running in Kubernetes
a handful of Laird gateways connecting via basicstation and 3 end devices (xdot based). It suddenly started happening after I upgraded to 3.9.0 from 3.8.7.

#### How do you propose to implement this?

There appears to be no check for a non-existent session in applicationserver.go:

line 1022:
```
var session *ttnpb.Session
	switch {
	case dev.Session != nil && bytes.Equal(dev.Session.SessionKeyID, msg.SessionKeyID):
		session = dev.Session
	case dev.PendingSession != nil && bytes.Equal(dev.PendingSession.SessionKeyID, msg.SessionKeyID):
		session = dev.PendingSession
	}
	if session.GetAppSKey() == nil {
		return errNoAppSKey.New()
	}
```

My naive fix, based on not understanding what could cause this is:
```
        if session == nil || session.GetAppSKey() == nil {
```

#### How do you propose to test this?

Works On My Machine 

#### Can you do this yourself and submit a Pull Request?

Yes. But only if you think it's a good idea and I haven't missed the point entirely.

adriansmares, 2020-08-10:
Thank you for your report !

This is indeed a real bug caused by the fact that in 99.9% of the cases, `Get*()` functions from our generated `protobuf` definitons work even if the object is `nil`. In fact, it should even now:
https://github.com/TheThingsNetwork/lorawan-stack/blob/f8a801a1e6d5cee53492bc4677542e3f7aaf292b/pkg/ttnpb/keys.pb.go#L238-L243

So why doesn't this happen then ? Because the session keys are embedded in the session:
https://github.com/TheThingsNetwork/lorawan-stack/blob/f8a801a1e6d5cee53492bc4677542e3f7aaf292b/pkg/ttnpb/end_device.pb.go#L66-L70

This means that in order to obtain the pointer to `SessionKeys` (in order to call `GetAppSKey`, which has a pointer receiver) the compiler must first dereference the `session`. But since the `session` is `nil`, it panics. I will check other places where we use the `GetAppSKey` and come back with a PR. 

cc @johanstokking

closed 2020-08-10 by adriansmares

johanstokking, 2020-08-10:
Thanks @veghead for reporting!

PR #3066 (2020-08-10): Fix session keys nil dereference

#### Summary

Closes #3053

#### Changes

- Check if the `session` is `nil`, since the `SessionKeys` are actually an embedded member of the session.

#### Testing

Unit testing.

##### Regressions

The intent of the `if` stays the same, no regressions should occur.

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards compatible with existing API, storage, configuration and CLI, according to the compatibility commitments in `README.md` for the chosen target branch.
- [ ]  [... cut]