TheThingsNetwork/lorawan-stack #3060: Build stand-alone C# SDK for TTES APIs

ymgupta, 2020-08-10:
#### Summary
A stand-alone C# SDK for TTES APIs, to integrate with an existing C# application
#### Why do we need this?
Having SDK in a programming language helps the developer to integrate quickly than integrating the REST APIs
#### What is already there? What do you see now?
Only JS SDK is present as of now 
#### What is missing? What do you want to see?
C# SDK along with SDK integration documentation
#### Environment
...
#### How do you propose to implement this?
...
#### How do you propose to test this?
...
#### Can you do this yourself and submit a Pull Request?
...

**Note:** 
Interested developers can 👍 the issue for the core team to prioritize

rvolosatovs, 2020-08-10:
@ymgupta please always set milestone

KrishnaIyer, 2025-04-30:
C# is not a commonly (if not ever) requested language to interact with The Things Stack. The community can pick this up if needed but this isn't something we are probably going to do.

closed 2025-04-30 by KrishnaIyer