TheThingsNetwork/lorawan-stack #3063: Documentation issue - migration from V2 to V3

laurensslats, 2020-08-10:
#### Summary

The below command should result in data not being processed in the V2 NS, but only in V3. However, it still ends up in our V2 network, and not in V3.

To reproduce:
1. Use the command as shown in documentation: `$ ttnctl devices convert-to-abp "device-id" --save-to-attribute "original-app-key"`
 [image]
2. Check the application in V2 and see data still being processed
[image]

#### Why do we need this ?
Make a seamless migration flow, from V2 to V3.

#### What is already there? What do you see now?
Docs: https://enterprise.thethingsstack.io/getting-started/migrating-from-v2/export-v2-devices/
Video: https://www.loom.com/share/20068ee431f34f40aadf83be280f2a90

#### What is missing? What do you want to see?
Find a new way on disabling devices to be routed in V2.

#### How do you propose to document this?
1. Define the right step
2. Document
3. Update video

#### Can you do this yourself and submit a Pull Request?
Need input from @neoaggelos

neoaggelos, 2020-08-10:
When migrating a device, the docs mention removing the `AppKey` of the device from v2. This means that the device can no longer join on the previous server. However, the active session of the device (session keys, frame counters) is not affected. This means that the device can obviously still receive send traffic to the v2.

I am not sure if v2 has a "disable" feature for devices. If not, then a solution would be either to clear the session keys of the device (from the v2), or even delete the device from v2 altogether.

laurensslats, 2020-08-14:
Let's keep in mind that customer are very scared to delete the session data of their devices (what if something goes wrong and 10k devices are bricked). Also considering that Packet Broker might not work in all cases, we need to add a clear step on Gateway Migration as a next steps after importing devices.json in TTS.

**Gateway migration (focussed on UDP packet forwarder)**

- Point to a new `server address`
- Update configuration. @htdvisser do people have to download the global_conf.json file when migration from V2 to V3? If yes, in what scenarios this is (not) required?

@neoaggelos why is the AppKey only added as an attribute, not added to the AppKey field in TTS?
[image]
[image]

htdvisser, 2020-08-17:
The configuration that you can download from The Things Stack is in most cases the same as the configuration that was previously in use on the gateway, so in most cases changing the address should be sufficient.

Of course, if gateway owners already need to spend time logging in to the gateway, it may be smart to recommend them to install something that let's them update the firmware/configuration remotely (like Balena or Basic Station), or at set up remote access (SSH) so that they can update the firmware/configuration remotely.

neoaggelos, 2020-08-18:
@laurensslats This is `prio/high`, but targetting Next Up.

What else is needed from me for this issue? Are there any actionable items? Please ask and assign accordingly.

closed 2020-08-24 by johanstokking