TheThingsNetwork/lorawan-stack #3067: ISM 2400 RX2 frequency should be 2.423 Ghz

Oliv4945, 2020-08-10:
#### Summary

Current ISM2400 RX2 frequency is `2422000000` ([here](https://github.com/TheThingsNetwork/lorawan-stack/blob/v3.9/pkg/band/ism_2400.go#L129)) , as written in last regional parameter draft ([v4](https://members.lora-alliance.org/communities/community-home/digestviewer/viewthread?GroupId=127&MessageKey=a8df33ba-803a-4298-a30e-2e2ab8e0f7ec&CommunityKey=6114003d-ea3c-4b44-bf19-3b0e56817ce5&tab=digestviewer) - page 13) it should be `2423000000`

Thank you :)

johanstokking, 2020-08-19:
@Oliv4945 we can't easily change the RX2 frequency as this will break existing devices; the RX2 frequency is not part of join-accept.

So we need:

1. To fix the frequency as 2.423 GHz is wrong and 2.422 GHz is right (i.e. this is a bug that we can patch)
2. Use a PHY version for this. This is not easy because this draft spec is not part of a RP001 or RP002 release
3. Introduce a new band for this that co-exists with `ISM_2400`. This becomes quite messy; PHY versions are meant for this
4. Don't care, treat ISM2400 as working draft (which is apparently what it is) and accept breaking RX2 for existing devices using an older draft. In that case, we can do this in upcoming 3.10 minor release (and not patch it in 3.9)

rvolosatovs, 2020-08-19:
Given our offline discussion, I guess we agree on 4. here ?

johanstokking, 2020-08-24:
Closed by https://github.com/TheThingsNetwork/lorawan-stack/pull/3074

closed 2020-08-24 by johanstokking

Oliv4945, 2020-09-01:
Hi,

I initially though that you can create a new frequency plan as the ISM2400 is currently named `LoRa 2.4 Ghz with 3 channels draft 2` but I now understand that it is not as easy so I agree with you that it is a working draft and it can be changed.

1. Do you have an other way than the changelog to warn current users of TTI instances ?
1. Frequency plan name `LoRa 2.4 Ghz with 3 channels draft 2` should now be renamed `LoRa 2.4 Ghz with 3 channels draft 4`

For information LoRa Basics Modem is [using 2.223 GHz](https://github.com/lorabasics/lorabasicsmodem/blob/9eac8cc24c59bd042a90e3de3c6d4b7a4cb66d96/lr1mac/src/smtc_real/src/region_ww2g4.h#L58) since its release so it might solve some user issues :)

PR #3074 (2020-08-20): Adjust ISM2400 frequencies according to v4 draft

#### Summary

Closes #3067 
cc @Oliv4945 

#### Changes

- Set RX2 frequency equal to what is demanded by the spec
- Set beacon/ping slot frequency to what is demanded by the spec

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards compatible with existing API, storage, configuration and CLI, according to the compatibility commitments in `README.md` for the chosen target branch.
- [x] Documentation: Relevant documentation is added or updated.
- [x] Chang [... cut]