TheThingsNetwork/lorawan-stack #3080: Device Template Converter stops after a single error

neoaggelos, 2020-08-11:
#### Summary

When importing multiple devices via `Import End Devices` from the Console, the import process stops if any template conversion fails.

This came up during development and testing in #3078 

#### Steps to Reproduce

1. Select `Import New Devices`
2. Try to import N devices, with one of them containing errors

#### What do you see now?

The Import process fails, ignoring the rest of the devices.

#### What do you want to see instead?

An option to continue importing the rest of the devices. If any errors occur in the process, then return a list of them.

#### Environment

v3.9.0 TTS

#### How do you propose to implement this?

In order to maintain backwards compatibility, we can add a new boolean option, e.g. `continue_on_error`, disabled by default. This can be a simple checkbox in the `Import End Devices` form. If the user does not enable it, then we maintain the current behaviour, i.e. fail if any device conversion fails. Otherwise, DTC keeps a list of any errors that may occur, continues importing the rest of the devices, and returns the list of errors in the end. I think it should not be _too_ hard to return a list of errors using the current response format, and we should definitely not break the API with this.

Adding discussion label, cc @johanstokking @htdvisser @laurensslats please give your opinion.

#### How do you propose to test this?

Unit testing should be simple

#### Can you do this yourself and submit a Pull Request?

yes

johanstokking, 2020-08-17:
> I think it should not be _too_ hard to return a list of errors using the current response format

Please elaborate, not breaking the (JSON) API. We should probably use warnings for this.

---

I don't think we should do this. That you ran into this during development and testing is understandable, but this shouldn't happen in production, unless there's something really wrong with the source file, which is a bad request that the client should fix.

neoaggelos, 2020-08-24:
Closing, let's reopen if this is something we want to revisit in the future.

closed 2020-08-24 by neoaggelos