TheThingsNetwork/lorawan-stack #3089: Unset uplink token if downlink is impossible

johanstokking, 2020-08-17:
#### Summary

Unset uplink token if downlink is impossible

#### Why do we need this?

- Preparation for https://github.com/TheThingsNetwork/lorawan-stack/issues/2708, specifically that upstream handling doesn't trigger downlink tasks
- Support Basic Station upstream messages with `xtime` of `0`: do handle (data uplink) messages, but don't trigger downlink tasks as timing is wrong

#### What is already there? What do you see now?

The uplink token is set if there is RX metadata.

#### What is missing? What do you want to see?

Conditionally set uplink token, not only by RX metadata availability, but also if the frontend decided that there's no downlink possible.

#### Environment

Since https://github.com/TheThingsNetwork/lorawan-stack/pull/3036

#### How do you propose to implement this?

- Make sure that `pkg/gatewayserver/io.(*Connection).HandleUp()` is aware of the downlink availability, potentially via a parameter
   - In that case, don't set an uplink token so NS won't try downlink
   - Also, don't sync the clock because the time is probably wrong
- Remove dropping join-requests and data uplink messages in Basic Station if `xtime` is `0` (see https://github.com/TheThingsNetwork/lorawan-stack/pull/3036/files#diff-8beea6478bacad4b4f77990a4719ea07R434 and https://github.com/TheThingsNetwork/lorawan-stack/pull/3036/files#diff-8beea6478bacad4b4f77990a4719ea07R456); indicate no downlink possibility instead

@rvolosatovs please confirm that NS doesn't start a downlink task if no (deduplicated) uplink messages have an uplink token. If this doesn't already happen, please do that or comment on #2708 that downlink tasks do not get started with no uplink tokens.

#### How do you propose to test this?

Unit testing

#### Can you do this yourself and submit a Pull Request?

Can review

rvolosatovs, 2020-08-18:
@rvolosatovs please confirm that NS doesn't start a downlink task if no (deduplicated) uplink messages have an uplink token. If this doesn't already happen, please do that or comment on #2708 that downlink tasks do not get started with no uplink tokens.

NS only adds downlink tasks when there is an available downlink path, so if there's no uplink tokens in metadata, there's not going to be a class A downlink attempt and no class B/C either, unless downlink paths are set on the downlinks themselves.

johanstokking, 2021-12-30:
This is already implemented for traffic coming in from Packet Broker.

Also, we have a downlink path constraint that is conveyed to NS, and if that is never, then downlink will not be attempted.

We need this indeed for traffic that is being received out-of-order or "late". We track that in https://github.com/TheThingsNetwork/lorawan-stack/issues/2708.

closed 2021-12-30 by johanstokking