TheThingsNetwork/lorawan-stack #3091: Update Go to 1.15

htdvisser, 2020-08-17:
#### Summary

This issue tracks the things we need to do before we can update Go to 1.15.

#### What is already there? What do you see now?

I started the `feature/go-1.15` branch. I already saw some failing tests locally, at least one of which related to Go 1.15:

```
 ERROR Could not start web server               endpoint=0.0.0.0:12391483 error=error:pkg/component:listen_endpoint (could not listen on `0.0.0.0:12391483` address) error_cause=error:pkg/errors:net_operation (listen tcp: address 12391483: invalid port) message=listen tcp: address 12391483: invalid port net=tcp op=listen temporary=false test_name=TestHTTP timeout=false
--- FAIL: TestHTTP (0.02s)
    doc.go:66:
        /Users/htdvisser/projects/lorawan-stack/pkg/component/component_test.go:159
        Expected: nil
        Actual:   'Get "https://localhost:9186/not%20found": x509: certificate relies on legacy Common Name field, use SANs or temporarily enable Common Name matching with GODEBUG=x509ignoreCN=0'
panic: runtime error: invalid memory address or nil pointer dereference [recovered]
	panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x10 pc=0x21bab38]

goroutine 163 [running]:
testing.tRunner.func1.1(0x231e5e0, 0x323b430)
	/usr/local/Cellar/go/1.15/libexec/src/testing/testing.go:1057 +0x30d
testing.tRunner.func1(0xc000603b00)
	/usr/local/Cellar/go/1.15/libexec/src/testing/testing.go:1060 +0x41a
panic(0x231e5e0, 0x323b430)
	/usr/local/Cellar/go/1.15/libexec/src/runtime/panic.go:969 +0x175
go.thethings.network/lorawan-stack/v3/pkg/component_test.TestHTTP(0xc000603b00)
	/Users/htdvisser/projects/lorawan-stack/pkg/component/component_test.go:160 +0xbf8
testing.tRunner(0xc000603b00, 0x25cab68)
	/usr/local/Cellar/go/1.15/libexec/src/testing/testing.go:1108 +0xef
created by testing.(*T).Run
	/usr/local/Cellar/go/1.15/libexec/src/testing/testing.go:1159 +0x386
FAIL	go.thethings.network/lorawan-stack/v3/pkg/component	1.852s
```

This one is for @johanstokking. I'll update the issue and assignees as we discover more issues.

johanstokking, 2020-08-24:
Blocked by #2975 as `otto` has issues with Go 1.15

rvolosatovs, 2020-09-02:
What's the status here? Is this still blocked?
I would like to use `t.Deadline()` from go 1.15 in #3052

johanstokking, 2020-09-02:
Not blocked anymore by `otto`

htdvisser, 2020-09-03:
@johanstokking please re-assign if there's anything left after you fix the testing certificates.

PR #3182 (2020-09-04): Update Go to 1.15

#### Summary

Closes #3091

#### Changes

- Update Go version in `go.mod` and GitHub Actions workflows
- Regenerate testing certificates to use SAN: https://golang.org/doc/go1.15#commonname

#### Testing

Locally

##### Regressions

Looking at https://golang.org/doc/go1.15 I don't see any potential regressions

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards compatible with existing API, storage, configuration and CLI, according to the compatibility commitments in `README.md` for t [... cut]
closed 2020-09-04 by johanstokking