TheThingsNetwork/lorawan-stack #3104: Incorrect frequency plan when exporting devices in TTN Stack V2

laurensslats, 2020-08-20:
## Summary
When exporting EU devices in V2, the frequency plan `EU_863_870` is set, not `EU_863_870_TTN`. As a result, SF12 for RX2 is used instead of SF9.

#### Steps to Reproduce
1. Export any EU device in TTN Stack V2, look at the frequency plan used

_Note: Haven't tested this for non-EU devices._

#### What do you see now?
Use the right Frequency plan ID.

#### What do you want to see instead?
AFAIK, TTN Stack V2 uses SF9 for RX2. If this is not set, the devices will miss some downlink windows.

#### How do you propose to implement this?

Option 1: updating `ttnctl`?
Option 2: import `EU_863_870` as `EU_863_870_TTN`

#### How do you propose to test this?
Run through a device migration.
Also test this with devices using other frequency plans

#### Can you do this yourself and submit a Pull Request?
Need help from @htdvisser / @neoaggelos 

Questions:
- @htdvisser Can the RX2 window be set in TTN Stack V2? If not (and V2 always uses SF9 for RX2), Option 2 (see: How do you propose to implement this?) is probably the best way forward.
- @neoaggelos is there an easy way to test if this issue occurs for other frequency plans?

htdvisser, 2020-08-20:
The RX2 data rate in v2 networks is determined by server configuration. All TTN deployments use SF9, but I know of some private deployments that use SF12. Note that this RX2 data rate difference is only an issue for ABP devices, OTAA devices will receive new settings on the next join.

neoaggelos, 2020-08-20:
The frequency plan ID is not determined by `ttnctl`, but rather passed by the user. [Quoting the docs on this](https://thethingsstack.io/getting-started/migrating-from-v2/export-v2-devices/#exporting-devices): 

>
> `$ ttnctl devices export "device-id" --frequency-plan-id EU_863_870 > device.json`
>  ...
>  NOTE: In the command above we used the EU_863_870 frequency plan. You will need to change this to the frequency plan corresponding to your region. You can find a list of supported frequency plan IDs in the lorawan-frequency-plans Github repository.

So, in order to use `EU_863_870_TTN`, one should simply do this instead:

```
$ ttnctl devices export "device-id" --frequency-plan-id EU_863_870_TTN > device.json`
```