TheThingsNetwork/lorawan-stack #3105: Registering device on network server returns error: Invalid value of field "ids.dev_addr"

Nihad32, 2020-08-20:
I'm trying to add device over HTTP POST, Adding on **identity server** over http seems to work.
Documentation says that after adding device to the identity server it needs to be added also on **network server** and **application server**. However when trying to add device to **network server** I get error: **Invalid value of field "ids.dev_addr"** 

I'm posting to _/api/v3/ns/applications/**application_id**/devices_
JSON i'm posting:
```
{
	"end_device": {
		"ids": {
			"device_id": "postdevicetest7",
			"dev_eui": "1111111111111117",
			"join_eui": "",
			"dev_addr": "77777777",
			"end_device_application_ids": {
				"application_id": "MyAppId"
			}
		},
		"name": "tmptest7",
		"description": "test desc7",
		"service_profile_id": null,
		"network_server_address": "ex.example.com",
		"application_server_address": "ex.example.com",
		"supports_class_b": false,
		"supports_class_c": true,
		"lorawan_version": "1.0.3",
		"lorawan_phy_version": "1.0",
		"frequency_plan_id": "EU_863_870",
		"in_frequency": 863,
		"max_frequency": 870,
		"supports_join": false,
		"resets_join_nonces": false,
		"last_dev_nonce": 0,
		"last_join_nonce": 0,
		"last_rj_count_0": 0,
		"last_rj_count_1": 0,
		"power_state": "POWER_UNKNOWN",
		"downlink_margin": 0,
		"multicast": false,
		"skip_payload_crypto": false
	},
	"field_mask": {
		"paths": ["frequency_plan_id", "lorawan_phy_version", "lorawan_version", "supports_join"]
	}
}
```
I tried leaving field empty, putting real device address, dummy hex ... I get the same error.
I tried adding **dev_addr** to the **field_mask** but then I get **_forbidden path(s) in field mask_**
I also tried the PUT method on /api/v3/ns/applications/**application_id**/devices/**device_id** and still got the same error

rvolosatovs, 2020-08-20:
There are multiple things wrong with the request and error you're seeing is just one of the issues.
You are creating an ABP device, however:
- `session.dev_addr` is missing (it must be equal to `ids.dev_addr`)
- Session keys are missing

Nihad32, 2020-08-21:
Thank you for your guidance, Error was misleading me.

I will try adding all fields that I'm missing and also add them to a field_mask.

closed 2020-08-24 by Nihad32