TheThingsNetwork/lorawan-stack #3112: Make documentation sidebar collapsible

nejraselimovic, 2020-08-20:
#### Summary
While navigating through the documentation site, subsections of the non-selected parent section should be hidden, while the subsections of the selected parent should be visible.

#### Why do we need this ?
As documentation grows, the sidebar navigation gets messier and this could help improving it.

#### What is already there? What do you see now?
We have this:
[image]

#### Can you do this yourself and submit a Pull Request?
@pgalic96 @kschiffer can you help?

johanstokking, 2020-08-24:
Is this actionable enough? Who is picking this up?

nejraselimovic, 2020-08-25:
> Is this actionable enough? Who is picking this up?

Did I assign the right people here?

johanstokking, 2020-08-25:
With three people assigned, who's responsible and who does what?

nejraselimovic, 2020-08-25:
> With three people assigned, who's responsible and who does what?

It's my mistake because I wasn't sure who can actually pull this off and ofc this discussion could be avoided if I asked who can do it before assigning them (will do in the future). I can help, but I don't know how to do this by myself. @kschiffer @pgalic96 can someone pick this one up please?

kschiffer, 2020-08-25:
Not your mistake. It's ok to assign more people when in doubt and it's the assignee's responsibility to narrow things down. I would say that @benolayinka is best for implementing this. It should be a relatively simple CSS job.

@benolayinka can you confirm and reassign if necessary then?

benolayinka, 2020-11-17:
Moved to https://github.com/TheThingsIndustries/lorawan-stack-docs/issues/97

closed 2020-11-17 by benolayinka