TheThingsNetwork/lorawan-stack #3124: Unclear if/how Location data of devices is used in console

laurensslats, 2020-08-24:
#### Summary
It's possible to add a location to a device, but it's not clear how/when this location is used and if this location can be set via other ways than adding the lat and lon in the console.

[image]

#### Why do we need this ?
It's not clear how this feature can be used. 

#### What is already there? What do you see now?
See img 👆 
Setting location via console.

#### What is missing? What do you want to see?
Documentation on:
- What are the sources for getting this location (only console, or also via the GPS coordinates of the devices, or via another way...) 
- If this is set, how can it be used? Only for reference in the console, or it is also added when initiating integrations?

#### How do you propose to document this?
1. Find out details from @htdvisser 
2. Write documentation by @benolayinka

johanstokking, 2020-08-24:
Setting the location is mostly used for in the Console.

Please file an additional issue for an "device overview" map where you can see all devices on the map, not only in a list.

References #2630; when we have geolocation there will be other sources that get filled out automatically.

benolayinka, 2020-11-11:
@johanstokking is there a way to move this issue to https://github.com/TheThingsIndustries/lorawan-stack-docs/tree/master/doc or do I have to close and recreate it?

bafonins, 2020-11-11:
@benolayinka close and recreate it

johanstokking, 2020-11-12:
> @benolayinka close and recreate it. there is no functionality to transfer issues between repositories as far as I know 🤷‍♂️ we usually just recreate those and mention the initial issue

There is but not across Github accounts.

closed 2020-11-17 by benolayinka