TheThingsNetwork/lorawan-stack #3145: Flaky test applicationserver linking_test

benolayinka, 2020-08-25:
#### Summary

https://github.com/TheThingsNetwork/lorawan-stack/pull/3139/checks?check_run_id=1026896468

johanstokking, 2020-08-25:
Could be resolved with https://github.com/TheThingsNetwork/lorawan-stack/issues/2312

KrishnaIyer, 2020-09-04:
Saw this again today
https://github.com/TheThingsNetwork/lorawan-stack/pull/3185/checks?check_run_id=1071013805

bafonins, 2020-09-09:
Same here

```
--- FAIL: TestLink (2.08s) 
--- FAIL: TestLink/InvalidAuth (1.03s) 
--- FAIL: TestLink/InvalidAuth/1 (0.51s) 
doc.go:66: 
/home/runner/work/lorawan-stack/lorawan-stack/pkg/applicationserver/linking_test.go:268 
Expected '<nil>' to NOT be nil (but it was)! 
linking_test.go:269: Unexpected error: <nil> 
FAIL 
coverage: 74.3% of statements
```

benolayinka, 2020-09-22:
@adriansmares where's the PR?

adriansmares, 2020-09-22:
https://github.com/TheThingsNetwork/lorawan-stack/pull/3263

rvolosatovs, 2020-10-09:
```
 --- FAIL: TestLink (2.09s)
    --- FAIL: TestLink/InvalidAuth (1.03s)
        --- FAIL: TestLink/InvalidAuth/1 (0.52s)
            doc.go:66: 
                /home/runner/work/lorawan-stack/lorawan-stack/pkg/applicationserver/linking_test.go:268
                Expected '<nil>' to NOT be nil (but it was)!
            linking_test.go:269: Unexpected error: <nil>
FAIL
coverage: 73.0% of statements
FAIL	go.thethings.network/lorawan-stack/v3/pkg/applicationserver	55.421s
```

johanstokking, 2020-10-10:
It's a bug but this doesn't happen too often and we're going to change this anyway (#3263) so let's not bother too much this month.

rvolosatovs, 2020-10-12:
Can we just remove the test already then?

johanstokking, 2020-10-12:
I'm also fine with that

htdvisser, 2020-12-28:
One more: https://github.com/TheThingsNetwork/lorawan-stack/runs/1617365245?check_suite_focus=true#step:14:34