TheThingsNetwork/lorawan-stack #3153: Create API key button with preset rights

johanstokking, 2020-08-26:
#### Summary

Generate API key button in Console with preset rights

#### Why do we need this?

In places where the user needs an API key with specific rights, to create one in one click.

#### What is already there? What do you see now?

Instructions to go to the API key page, create a new key and select specific rights, copy the key and go back to where you need them.

#### What is missing? What do you want to see?

A button "Create API key" in the Console that immediately opens the popup where the user can copy the key. The API key name and rights are passed via properties of the button.

This would be useful in the MQTT server integration screen, but also in the AWS IoT Core screen of TTES.

#### How do you propose to implement this?

Not sure, probably a new button component?

@kschiffer to think about the UX.

#### How do you propose to test this?

Test locally

#### Can you do this yourself and submit a Pull Request?

Can review user experience but not the code

Nihad32, 2020-08-27:
Hi,

is that key same as **User API key** mentioned [here](https://thethingsstack.io/reference/api/authentication/) ? Can that key be used for http authentication ? 

Sorry if this is the wrong place to ask, I'm struggling to figure out where to create one auth key that can to everything listed [here](https://thethingsstack.io/reference/api/)

kschiffer, 2021-09-14:
This is already implemented. Both in MQTT integration and AWS IoT core, it is possible to click the "Generate API Key" button to generate the key right away. @johanstokking please reopen if anything else is necessary.

closed 2021-09-14 by kschiffer