TheThingsNetwork/lorawan-stack #3167: Improve documentation for EndDevice registry methods and services

benolayinka, 2020-08-31:
#### Summary

Subtask of #252

From the umbrella issue:

- Improved documentation on methods (for example, explaining that As.SetLink does not return whether the actual linking is successful, perhaps explaining how many resources a method call consumes)
- Improved documentation on services (for example, being more clear about which components expose the services, which fields are accepted by the different EndDevice registries).

#### What is already there? What do you see now?

https://github.com/TheThingsNetwork/lorawan-stack/blob/v3.9/api/end_device.proto#L140

#### What is missing? What do you want to see?

See https://github.com/TheThingsNetwork/lorawan-stack/pull/3015/files for reference

#### Can you do this yourself and submit a Pull Request?

No. @rvolosatovs please take a little time and just get this out of the way quickly

johanstokking, 2020-11-12:
> No. @rvolosatovs please take a little time and just get this out of the way quickly

@benolayinka you can do this yourself. Please let us know if you need help!

benolayinka, 2020-11-18:
Reassigning to @rvolosatovs https://github.com/TheThingsNetwork/lorawan-stack/issues/252#issuecomment-729164178

closed 2021-02-24 by benolayinka