TheThingsNetwork/lorawan-stack #3177: Authenticate gateways with V2 access key

johanstokking, 2020-09-03:
#### Summary

Authenticate gateways with V2 access key

#### Why do we need this?

To authenticate gateways that are currently connected to V2 but are being migrated to V3, without changing their gateway access key to a V3 API key.

#### What is already there? What do you see now?

Gateway Server optionally supports unauthenticated gateways, or authenticates via V3 API key.

#### What is missing? What do you want to see?

Authenticate optionally via V2 access key that is stored in a secret.

#### How do you propose to implement this?

- Add V2 access key as gateway entity secret (blocked by https://github.com/TheThingsIndustries/lorawan-stack/pull/2312, becomes open source)
- Check if the gateway authenticates with a V2 access key. Note that this prefix is configurable. This should probably only be supported in MQTT

#### How do you propose to test this?

Integration testing

#### Can you do this yourself and submit a Pull Request?

Can review

johanstokking, 2020-09-03:
@KrishnaIyer please post diff with API and DB model for the secret before implementation

KrishnaIyer, 2020-09-07:
I'm looking at this now and this might be a dumb question but; if someone needs to change the endpoint on the gateway for migration, they might as well change the API Key right?

johanstokking, 2020-09-08:
This is for in-place upgrades, from V2 Bridge to V3 Gateway Server, i.e. for forwarding purposes. For example, V2 private network clusters that get upgraded to V3, either to fully fledged V3 clusters, or V3 GS + PBA.

johanstokking, 2020-11-12:
@KrishnaIyer changed this from a gateway secret in IS to authenticating the gateway on the V2 Account Server.

KrishnaIyer, 2020-11-12:
😭

KrishnaIyer, 2020-11-12:
So v3 GS talks to V2 AS?

johanstokking, 2020-11-12:
Yes.

KrishnaIyer, 2020-11-12:
Ok replaced by https://github.com/TheThingsIndustries/lorawan-stack/issues/2428

closed 2020-11-12 by KrishnaIyer