TheThingsNetwork/lorawan-stack #3202: Automated end-to-end testing

johanstokking, 2020-09-09:
#### Summary

Automated end-to-end testing.

#### Why do we need this?

Regression testing in CI.

#### What is already there? What do you see now?

- Automated frontend testing: https://github.com/TheThingsNetwork/lorawan-stack/issues/2605
- CLI for simulating join-requests and uplink messages

#### What is missing? What do you want to see?

Scripting common use cases, i.e. creating a gateway, application and device in the Console, sending a simulated join-request via CLI, asserting events displayed in Console, scheduling downlink message, asserting that the downlink message arrives as response to a simulated uplink message via CLI, etc.

#### Environment

CI

#### How do you propose to implement this?

Not sure, please provide input

#### How do you propose to test this?

CI

#### Can you do this yourself and submit a Pull Request?

No

rvolosatovs, 2021-01-04:
Adding me and @KrishnaIyer as we discussed this on last sync

rvolosatovs, 2021-01-04:
Refs #3606

KrishnaIyer, 2021-03-31:
Refs https://github.com/TheThingsIndustries/lorawan-stack/issues/2652

KrishnaIyer, 2021-03-31:
This needs to get moving. I've bumped https://github.com/TheThingsNetwork/lorawan-stack/issues/3202 to `v3.12.1`. I'll do the necessary groundwork and revert.

I'll then create issues per function and assign members from this list so we can make some progres.

kschiffer, 2021-03-31:
Key issue to move on with this is this one: #3611 

I started working on that already but had to stop for other issues.

johanstokking, 2021-06-28:
What is the status here?

KrishnaIyer, 2021-09-09:
Replaced by https://github.com/TheThingsIndustries/lorawan-stack/issues/2742

closed 2021-09-09 by KrishnaIyer