TheThingsNetwork/lorawan-stack #3209: Gateway server issues duplicate drop uplink message events

kschiffer, 2020-09-10:
#### Summary
The gateway server occasionally issues the exact same `gs.up.drop` event two times, even with the same timestamp. This should not happen.

References #3098, #580

#### Steps to Reproduce
1. Subscribe to a gateway stream with sufficient traffic
2. Wait
3. See identical `gs.up.drop` events appearing

Example event arriving two times:
```
{
  "name": "gs.up.drop",
  "time": "2020-09-10T07:45:43.830867548Z",
  "identifiers": [
    {
      "gateway_ids": {
         …
      }
    }
  ],
  "data": {
    "@type": "type.googleapis.com/ttn.lorawan.v3.ErrorDetails",
    "namespace": "pkg/encoding/lorawan",
    "name": "field_length_with_multiple_choices",
    "message_format": "`{lorawan_field}` length `{value}` should be equal to `{expected_1}` or `{expected_2}`",
    "attributes": {
      "expected_1": 19,
      "expected_2": 24,
      "lorawan_field": "RejoinRequestPHYPayload",
      "value": 206
    },
    "code": 3
  },
  "correlation_ids": [
    "gs:conn:01EHCKG4RX6XM6ZKWRC2S3NR2H",
    "gs:uplink:01EHVFMDWNX4G7F6PPA8E8HRYF"
  ],
  "origin": "redacted",
  "context": {
    "tenant-id": "CgN0dGk="
  },
  "visibility": {
    "rights": [
      "RIGHT_GATEWAY_TRAFFIC_READ"
    ]
  }

```
#### What do you see now?
Duplicate events.

#### What do you want to see instead?
Duplicate events being discarded and not broadcasted to the event stream.

#### Environment
v3.9.x

#### How do you propose to implement this?
🤷 

#### How do you propose to test this?
Unit tests

#### Can you do this yourself and submit a Pull Request?
No. @johanstokking

johanstokking, 2020-09-10:
This is two goroutines for multiple upstreams that are handling the message concurrently. Two options: #580 (clean option) or add a correlation ID for the upstream (work-around for these particular event pairs).

johanstokking, 2020-09-17:
Closed with #580

closed 2020-09-17 by johanstokking