TheThingsNetwork/lorawan-stack #3221: Make events pub/sub format configurable

adriansmares, 2020-09-11:
#### Summary

The event backends should have a configurable format.

#### Why do we need this?

In order to lower runtime costs on encoding/decoding JSON, and to lower the amount of network traffic generated by our events.

#### What is already there? What do you see now?

Marshal/unmarshal using `encoding/json`. `go-cloud` offers internally `protobuf` usage, but this is not available over settings.

#### What is missing? What do you want to see?

Configurable marshaling, between `json` and `protobuf`.

#### Environment

`v3.9.2`

#### How do you propose to implement this?

- Add a configuration field to backends for `format` 
- Make this format be part of the factory function.
- Use the selected format at runtime.

#### How do you propose to test this?

Check using an appropiate pub/sub client if the format used is correct.

#### Can you do this yourself and submit a Pull Request?

Yes. cc @htdvisser

htdvisser, 2020-09-14:
When we initially built the events system back in 2018, we implemented the "internal" and "redis" backends as a reference only. The idea was that we'd migrate to a different backend such as RabbitMQ, Cloud Pub/Sub or NSQ/NATS for more "serious" deployments.

The "cloud" backend already uses protobuf as a default (and has support for json as well). 

And to be honest, I don't think the biggest scalability problem is the encoding (JSON vs Protobuf), but rather the fact that all events consumers (i.e. all console instances) subscribe to the firehose of all events and filter them locally, instead of routing events on the pub/sub broker or through some separate router (such as in Packet Broker).

adriansmares, 2020-09-14:
> When we initially built the events system back in 2018, we implemented the "internal" and "redis" backends as a reference only. The idea was that we'd migrate to a different backend such as [...]

That's most likely correct - but keep in mind that we had some interesting performance gains by simply changing the JSON library in some cases - I'd really like to see if `protobuf` would be better, at least initially.

But indeed - the real problem is the routing is way too trivial at the moment, so refactoring the events backend is inevitable - especially since the Event Server will need `exactly-once` semantics for HA (which we don't have right now).

htdvisser, 2020-09-14:
If we make the encoding configurable, we'd need to somehow detect the encoding when receiving events from Redis (not all components in a cluster may be deployed with the same configuration). Unlike the `cloud` backend, the Redis backend doesn't already have headers, so we'd have to come up with a different way to do this. Checking if the first bytes are `{"` might work, but it's probably better to be a bit more explicit by prefixing the event data with `json:` and `proto:` (and fall back to json if there is no such prefix).

adriansmares, 2020-09-14:
> If we make the encoding configurable, we'd need to somehow detect the encoding when receiving events from Redis (not all components in a cluster may be deployed with the same configuration). Unlike [...]

If all of the components share the same broker settings, is it completely unlikely that they will share the same format settings ? I'm ok with the prefixes too, but I'm a bit confused on why this content type problem arises in the first place.

htdvisser, 2020-09-14:
Different instances in a cluster are not guaranteed to run with the same configuration. This shouldn't cause errors on other instances in the same cluster.

htdvisser, 2021-09-23:
I don't think we need to do this anymore. The default format is already protobuf, so I don't think we need to make the format configurable anymore.

closed 2021-09-23 by htdvisser