TheThingsNetwork/lorawan-stack #3223: Add CUPS address and LNS address to gateway overview

benolayinka, 2020-09-14:
#### Summary

Add CUPS address and LNS address to gateway overview

#### Why do we need this?

Hugely simplify finding this information. Currently, LNS is by default `wss://server_address:8887` and CUPS is `https://server_address:443` which there is no way to figure out without reading BOTH the gateway overview and the LBS docs 

#### What is already there? What do you see now?

[image]

#### What is missing? What do you want to see?

Fields with CUPS / LNS server addresses

#### Can you do this yourself and submit a Pull Request?

Yeah but only if @bafonins or @kschiffer really really don't want to

KrishnaIyer, 2020-09-14:
Please hold this on https://github.com/TheThingsIndustries/lorawan-stack/issues/2088

kschiffer, 2021-06-01:
@benolayinka please consider adding this yourself if possible and please set an appropriate milestone in case. Otherwise reassign me.

benolayinka, 2021-07-12:
Giving this one back to you @kschiffer

benolayinka, 2021-07-12:
> Giving this one back to you @kschiffer

Gateway server address, LNS server address, and CUPS server address are used mutually exclusively, so this could also be done as a tab next to the gateway box to switch between those addresses