TheThingsNetwork/lorawan-stack #3259: Downlink pushing broken by #3190

rvolosatovs, 2020-09-21:
#### Summary

Downlink pushing broken by #3190

#### Steps to Reproduce

```
$ git checkout 87f6a5b55
$ cli device downlink push test-app2 test-dev-c --f_port 0x22 --frm_payload 0x000e
$ git checkout 954534b86
$ cli device downlink push test-app2 test-dev-c --f_port 0x22 --frm_payload 0x000e 
```

#### What do you see now?

```
$ gn 87f6a5b55
HEAD is now at 87f6a5b55 Merge pull request #3190 from TheThingsNetwork/feature/2312-as-link-removal
$ cli device downlink push test-app2 test-dev-c --f_port 0x22 --frm_payload 0x000e
 DEBUG Using access token (valid until 3:05PM) 
 DEBUG ccResolverWrapper: sending update to cc: {[{localhost:8884  <nil> 0 <nil>}] <nil> <nil>}
 DEBUG Channel switches to new LB policy "pick_first"
 DEBUG Subchannel Connectivity change to CONNECTING
 DEBUG Subchannel picks a new address "localhost:8884" to connect
 DEBUG pickfirstBalancer: HandleSubConnStateChange: 0xc000c85630, {CONNECTING <nil>}
 DEBUG Channel Connectivity change to CONNECTING
 DEBUG Subchannel Connectivity change to READY 
 DEBUG pickfirstBalancer: HandleSubConnStateChange: 0xc000c85630, {READY <nil>}
 DEBUG Channel Connectivity change to READY    
  WARN Finished unary call                      duration=26.771746ms error=rpc error: code = PermissionDenied desc = error:pkg/auth/rights:insufficient_application_rights (insufficient rights for application `test-app2`) error_correlation_id=7ee6f5a9e3c2450e9879b3c3b233d0d2 error_name=insufficient_application_rights error_namespace=pkg/auth/rights grpc_code=PermissionDenied grpc_method=DownlinkQueuePush grpc_service=ttn.lorawan.v3.AppAs namespace=grpc
error:pkg/auth/rights:insufficient_application_rights (insufficient rights for application `test-app2`)
    uid=test-app2
    correlation_id=7ee6f5a9e3c2450e9879b3c3b233d0d2
exit status 255
$ gn 954534b86
Previous HEAD position was 87f6a5b55 Merge pull request #3190 from TheThingsNetwork/feature/2312-as-link-removal
HEAD is now at 954534b86 Merge pull request #3249 from rvolosatovs/merge/3.9
$ cli device downlink push test-app2 test-dev-c --f_port 0x22 --frm_payload 0x000e 
 DEBUG Using access token (valid until 3:05PM) 
 DEBUG ccResolverWrapper: sending update to cc: {[{localhost:8884  <nil> 0 <nil>}] <nil> <nil>}
 DEBUG Channel switches to new LB policy "pick_first"
 DEBUG Subchannel Connectivity change to CONNECTING
 DEBUG Subchannel picks a new address "localhost:8884" to connect
 DEBUG pickfirstBalancer: HandleSubConnStateChange: 0xc000d9a7a0, {CONNECTING <nil>}
 DEBUG Channel Connectivity change to CONNECTING
 DEBUG Subchannel Connectivity change to READY 
 DEBUG pickfirstBalancer: HandleSubConnStateChange: 0xc000d9a7a0, {READY <nil>}
 DEBUG Channel Connectivity change to READY    
 DEBUG Finished unary call                      duration=347.673571ms grpc_method=DownlinkQueuePush grpc_service=ttn.lorawan.v3.AppAs namespace=grpc
 DEBUG Channel Connectivity change to SHUTDOWN 
 DEBUG Subchannel Connectivity change to SHUTDOWN
```

#### What do you want to see instead?

```
$ gn 87f6a5b55
HEAD is now at 87f6a5b55 Merge pull request #3190 from TheThingsNetwork/feature/2312-as-link-removal
$ cli device downlink push test-app2 test-dev-c --f_port 0x22 --frm_payload 0x000e
 DEBUG Using access token (valid until 3:05PM) 
 DEBUG ccResolverWrapper: sending update to cc: {[{localhost:8884  <nil> 0 <nil>}] <nil> <nil>}
 DEBUG Channel switches to new LB policy "pick_first"
 DEBUG Subchannel Connectivity change to CONNECTING
 DEBUG Subchannel picks a new address "localhost:8884" to connect
 DEBUG pickfirstBalancer: HandleSubConnStateChange: 0xc000d9a7a0, {CONNECTING <nil>}
 DEBUG Channel Connectivity change to CONNECTING
 DEBUG Subchannel Connectivity change to READY 
 DEBUG pickfirstBalancer: HandleSubConnStateChange: 0xc000d9a7a0, {READY <nil>}
 DEBUG Channel Connectivity change to READY    
 DEBUG Finished unary call                      duration=347.673571ms grpc_method=DownlinkQueuePush grpc_service=ttn.lorawan.v3.AppAs namespace=grpc
 DEBUG Channel Connectivity change to SHUTDOWN 
 DEBUG Subchannel Connectivity change to SHUTDOWN
```

#### How do you propose to implement this?

No idea

#### How do you propose to test this?

See above

#### Can you do this yourself and submit a Pull Request?

I prefer not to

rvolosatovs, 2020-09-21:
Setting `prio/medium`, because being able to schedule application downlinks is a core feature of our stack

adriansmares, 2020-09-21:
I shouldn't have merged this until NS was done. Opening the a revert PR and creating a PR for #3190

rvolosatovs, 2020-09-21:
How is the NS related here?
If NS should just check different set of rights of push, I can open a quickfix PR for that maybe?

adriansmares, 2020-09-21:
> How is the NS related here?
> If NS should just check different set of rights on push, I can open a quickfix PR for that maybe?

Since there is no more linking, `Push` and `Replace` use cluster authentication instead of the the link API key.

I see two options:
- Revert #3190 completely - can be done by https://github.com/TheThingsNetwork/lorawan-stack/pull/3260
- Update the rights check in the NS - not hard, but since the NS doesn't link anything at the moment, uplinks won't work.

Solution 1 basically reverts `v3.10` to a 'working state' while solution 2 only makes downlink working again (but you don't have invalidations, so working is relative here).

adriansmares, 2020-09-21:
Closed by #3260

closed 2020-09-21 by adriansmares