TheThingsNetwork/lorawan-stack #3261: Provide a static HTML template that renders errors outside of the console/oauth "namespace"

kschiffer, 2020-09-21:
#### Summary
Following up from https://github.com/TheThingsNetwork/lorawan-stack/issues/233 we still need a Go HTML template that will render a generic and static error page for all errors that are not in Console or OAuth app context. See also: https://github.com/TheThingsIndustries/lorawan-stack/issues/1315#issuecomment-440656553

#### Why do we need this?
To display proper error pages for backend errors that are not in the scope of Console or OAuth app. E.g. https://github.com/TheThingsIndustries/lorawan-stack/issues/2268

#### What is already there? What do you see now?
Error page designs and error boundaries that render error views for routes that are handled by our frontend react apps.

#### What is missing? What do you want to see?

Templates that render static HTML/CSS for any route that is not handled by our frontend react apps.

#### How do you propose to implement this?
Implementing the design that we currently use for the react apps as static HTML/CSS template that renders for error responses.

#### How do you propose to test this?
Manual.

#### Can you do this yourself and submit a Pull Request?
Yes but anyone please feel free to pick up.

kschiffer, 2021-04-01:
Blocking: https://github.com/TheThingsIndustries/lorawan-stack/issues/1813

kschiffer, 2021-04-01:
@pierrephz We need a generic error page for TTS. I think you designed something like that already a while back. Can you post a recent design here? Such design should be pretty simple and contain:

*   Text for the error title
*   Text for resolution instructions
*   Some icon/visual signalling "error", with proper TTS branding
*   A link leading back to a homepage
*   A link to documentation
*   A link to support (Enterprise)

NicolasMrad, 2021-09-16:
@kschiffer can you please advise on how this error page will look like?

kschiffer, 2021-09-16:
Yes, so we should essentially use the same design that we currently use for the react handled errors (`<FullViewError />`):

[image]

We can use a simplified footer without the buttons on the right. Also, support buttons cannot be rendered in these cases, since the config is only available in the Account app / Console.

PR #4884 (2021-12-08): Non-react route error template

#### Summary

Closes #3261 

#### Changes

- Add Go error template
- Move assets necessary for rendering the error template to a separate directory used by the template
- Fix bug where not found routes were not handled properly in the main router

#### Testing

...

##### Regressions

None

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [ ] Compatibility: The changes are backwards compatible with existing API, storage, configuration and CLI, according to the compatibility commitments in `README.md` for the chose [... cut]
closed 2021-12-08 by pgalic96