TheThingsNetwork/lorawan-stack #3278: Raspberry Pi-4 installation - working or not ?

ecities, 2020-09-24:
#### Summary

**...Hi, for a few days we are trying to install and run TTN stack on RPI-4 buster without success.
We tried docker installation - unsucessfully.
We found .deb installation files and tried to install it.
Is there any other method or documentation "from scratch" how to run TTN Stack on RPI-4 buster.**
How to move forward - is there some tricky way to do it - or it doesn't work at all?

BR, Rob

#### Why do we need this ?

Use ttn stack on most common microcomputer board
...

#### What is already there? What do you see now?

...

#### What is missing? What do you want to see?

...

#### How do you propose to document this?

...

#### Can you do this yourself and submit a Pull Request?

...

johanstokking, 2020-10-05:
RPi installations are currently not documented. We do have binaries available for this target, see the `arm64` binaries you can download here: https://github.com/TheThingsNetwork/lorawan-stack/releases/tag/v3.9.4. You would use systemd to create a service for the RPi.

We're happy to see a pull request for the documentation for RPi users.

kschiffer, 2021-08-16:
@htdvisser since you made the webinar about it. Is there anything we can/should add from that?

htdvisser, 2021-08-16:
Not really. Our documentation doesn't need to cover setting up the Raspberry Pi itself, and the part where I install The Things Stack just follows the Docker instructions that we already have in the docs.

If anyone needs step-by-step instructions for the RPi specifically, they can watch [my conference talk](https://www.youtube.com/watch?v=7UlqLtl8GcU) or read [the transcript](https://www.thethingsnetwork.org/article/deploy-the-things-stack-in-your-local-network).

closed 2021-08-16 by htdvisser