TheThingsNetwork/lorawan-stack #3280: Make GCS API available via gRPC

kschiffer, 2020-09-24:
#### Summary
The Gateway Configuration Server currently does not use gRPC so it's API is plain HTML. In the JS SDK we somewhat rely on functionality being available as RPC so adding GCS functionality requires us to break out of our usual implementation.

#### Why do we need this?
- Retain implementation consistency in the JS SDK
- Avoid harcoded API routes and arbitrary implementations

#### What is already there? What do you see now?
gRPC API of all other stack components

#### What is missing? What do you want to see?
gRPC API for the GCS

#### How do you propose to implement this?
Likely recreate the implementation for the other stack components.

#### How do you propose to test this?
Unit tests.

#### Can you do this yourself and submit a Pull Request?
No, @KrishnaIyer @adriansmares please reassign as appropriate.

KrishnaIyer, 2021-12-01:
This is a nice technical debt thing to do, though atm the GCS doesn't expose any functionality to console.

KrishnaIyer, 2021-12-16:
GCS doesn't actually expose any functionality to the console. Do you use the GCS APIs in the console/SDK somewhere @kschiffer?

adriansmares, 2021-12-16:
The download gateway configuration button is from the GCS.

nicholaspcr, 2022-02-07:
Comment left by @htdvisser on PR and is a good question for the discussion
>I wasn't involved in the original discussion about building this feature, so this may be a bit late, but still. I don't think that "in the JS SDK we expect everything to be gRPC" is a good argument for forcing something that should be a simple file download into a gRPC API. Are we sure we really need to do this here? Why can't the SDK just support the file downloads that we already have?

From the @kschiffer description it seems to be just a question of breaking out of the standard implementation but might be best to have his response on the subject

kschiffer, 2022-02-07:
It's not like it would not work the way it is. It just forces us to [hardcode the API endpoint in the SDK](https://github.com/TheThingsNetwork/lorawan-stack/blob/v3.18/sdk/js/src/service/gateways.js#L256-L259) and also to break out of our regular HTTP request logic, while everywhere else we rely on the gRPC API definition to source API paths. I thought it was odd that the configuration download is not part of the gRPC API, like everything else. I don't have any very strong feelings about this otherwise than practicality and retaining consistency in the JS SDK.

johanstokking, 2022-02-10:
Actually I think in this case it makes sense to have a gRPC API to generate configuration and return that.

The reason that we started with the HTTP API is that gateways were able to download their own configuration file directly, if they are configured with their gateway ID and a gateway API key. We did not make it HTTP because the result happens to be a JSON file in all of the current cases.

PR #5174 (2022-02-28): Add gRPC implementation to GCS API

#### Summary

Closes #3280

#### Changes

- Add proto definition of GCS service
- Add grpc implementation of GCS service
- Add gRPC tests

#### Testing

Unit tests and manually testing by downloading the global config in the gateway page of console.

##### Regressions

The gRPC service is adjacent to the http routes that already exists, meaning that the route generated by the `grpc-gateway` is distinct from the previous routes. 

#### Notes for Reviewers

After the discussion in this PR thread, it was decided to create the gRPC http routes alongside with the [... cut]
closed 2022-02-28 by nicholaspcr