TheThingsNetwork/lorawan-stack #3287: Network server fails to schedule downlink on MP Packet forwarder (MQTT) based gateway

ymgupta, 2020-09-28:
#### Summary
Join accept is not being scheduled by the network server in the case of MQTT Gateway.
Ref: [#137](https://github.com/TheThingsIndustries/lorawan-stack-support/issues/137)
#### Steps to Reproduce
1. Register and connect an MP packet-forwarder enabled gateway to The Things Stack
2. Register a device
3. Activate via OTAA
#### What do you see now?
The following events are not triggered:

- `ns.down.join.schedule.attempt`
- `gs.down.send`
- `ns.down.join.schedule.success`

#### What do you want to see instead?
Successful joins and uplinks.
#### Environment
v3.9.4
#### How do you propose to implement this?
...
#### How do you propose to test this?
...
#### Can you do this yourself and submit a Pull Request?
Trying to debug

KrishnaIyer, 2020-09-30:
This issue heading/summary is quite misleading. So far **only** MP Packet forwarder fails. TTKG (which also works on MQTT) has no issues from my tests.
This is something to fix regardless.

KrishnaIyer, 2020-10-06:
closed via #3313

closed 2020-10-06 by KrishnaIyer