TheThingsNetwork/lorawan-stack #3311: Show advanced and interactive event details in the Console

kschiffer, 2020-10-01:
#### Summary
The Console currently only shows the raw event JSON in the event details panel (sidebar). We could provide a better experience here by facilitating common actions related to the event payload.

#### Why do we need this?
To further improve the UX around event and activity monitoring, facilitate troubleshooting and diagnosing of issues

#### What is already there? What do you see now?
- Showing the raw event JSON payload in the event sidebar
- Showing data previews per event data type

#### What is missing? What do you want to see?
- Mapping `devAddr`'s back to `devEUIs` for owned end devices (in gateway event streams)
- Cross-link owned entities (devices, gateways, applications) if possible
- Show data visualizations for gateway metrics, `rx_metadata` and similar data
- Show correlated events (using `correlation_ids`)
- Happy to hear more suggestions, cc @bafonins @johanstokking @htdvisser @ajovertoom 

#### How do you propose to implement this?
I'd like to collect suggestions and ideas first, then we can go ahead and maybe create individual more actionable issues.

#### How do you propose to test this?
This can be tested using storybook using realistic event data. Later on we can also run e2e tests on these.

#### Can you do this yourself and submit a Pull Request?
Yes.

kschiffer, 2020-11-16:
> Cross-link owned entities (devices, gateways, applications) if possible

@bafonins this touches on the linking feature discussed today in the Console meeting. It's part of an overall UX enhancement plan regarding event views in the Console. I wanted to collect ideas here and then follow up with specifications (wireframes) for individual actionable issues.

htdvisser, 2020-11-17:
> Mapping devAddr's back to devEUIs for owned end devices (in gateway event streams)

Not easy to do, and it's not guaranteed to be a unique mapping

> Cross-link owned entities (devices, gateways, applications) if possible

I think this needs some more explanation

> Show data visualizations for gateway metrics, rx_metadata and similar data

You mean something like "signal strength bars" for RSSI and SNR values or something?

kschiffer, 2020-11-17:
> Not easy to do, and it's not guaranteed to be a unique mapping

I would envision some backend API that lists owned devices by the sessions currently assigned DevAddr. There would be a very slight chance of an address overlap but I think we can take that risk if we communicate that.

> Cross-link owned entities (devices, gateways, applications) if possible

For troubleshooting purposes, it is very helpful to be able to jump from a gateway message event to the exact end device using the DevAddr, as long as such mapping can be made (see above).

> Show data visualizations for gateway metrics, rx\_metadata and similar data

I need to conduct some more research on gateways here to propose some concrete features, but signal strength bars would be a good start. Also, a visualization of the channel utilization would be helpful.

htdvisser, 2020-11-17:
> I would envision some backend API that lists owned devices by the sessions currently assigned DevAddr.

I figured, and that's exactly what makes it so difficult, because that's not how our database is modeled.

> There would be a very slight chance of an address overlap but I think we can take that risk if we communicate that.

That is absolutely not true. We (will) have hundreds of devices with the same DevAddr. In applications with thousands of devices there is an extremely high chance of multiple devices having the same DevAddr.

kschiffer, 2020-11-17:
> I figured, and that's exactly what makes it so difficult, because that's not how our database is modeled.

Difficult as in impossible?

> That is absolutely not true. We (will) have hundreds of devices with the same DevAddr. In applications with thousands of devices there is an extremely high chance of multiple devices having the same DevAddr.

Alright, had a misconception there. Would there be no way to reliably map the messages back to the owned end device? E.g. using the MIC and DevAddr?

htdvisser, 2020-11-17:
The only reasonable way would be by using the correlation IDs to find related events and use the device identifiers in those.

kschiffer, 2021-02-16:
Blocked on #3817

KrishnaIyer, 2024-10-14:
There are too many things in this issue varying degrees of complexity. I'll close this and open separate items if necessary

closed 2024-10-14 by KrishnaIyer