TheThingsNetwork/lorawan-stack #3327: Make gateways searcheable by gateway eui

kschiffer, 2020-10-07:
#### Summary
It should be possible to search gateways by their EUIs.
 
#### Why do we need this?
To help users find gateways that they don't know the gateway id of.

#### What is already there? What do you see now?
- Generic search API for entities
- End device search API

#### What is missing? What do you want to see?
- A specific gateway search API similar to the end device search API
- Wiring in the console to make use of such API

#### Environment
v3.10.x

#### How do you propose to implement this?
1. Add API, complementary to the [end device search API](https://github.com/TheThingsNetwork/lorawan-stack/blob/v3.10/api/search_services.proto#L92). cc @htdvisser 
2. Adjust console logic accordingly cc @pgalic96 

#### How do you propose to test this?
Unit tests as well as e2e tests in the console.

htdvisser, 2021-03-31:
The API was added in https://github.com/TheThingsNetwork/lorawan-stack/pull/3863, so this only involves the Console (and perhaps JS SDK) now.

pgalic96, 2021-11-09:
Closed because separate issues are merged together #1825

closed 2021-11-09 by pgalic96