TheThingsNetwork/lorawan-stack #3332: Determine device's operating band at join time

rvolosatovs, 2020-10-07:
#### Summary

Determine device's operating band at join time

#### Why do we need this?

A device may be able to operate in different regions and so Network Server could determine the frequency plan and band to use from the transmission settings of the join-request (or even first uplink for ABP)

#### What is already there? What do you see now?

A singular frequency plan configured per-device. Network Server uses that always.

#### What is missing? What do you want to see?

Determine the frequency plan to use for device depending on the join transmission settings.

#### How do you propose to implement this?

Allow several frequency plans defined per-device from different bands with no join frequency overlap (i.e. the join frequencies should uniquely identify the frequency plan given the frequency plan set configured for the device, we may use data rate here as well). This way depending on the frequency used for the join-request NS can determine the frequency plan to use for the session.
The ABP implementation can come later and would need more elaborate handling. Perhaps we should not do it at all, in any case we should try this with OTAA first.

#### How do you propose to test this?

Provision 2 UNOs with same identifiers/keys, but different bands(e.g. US915 and EU868) and join.

#### Can you do this yourself and submit a Pull Request?

yes

rvolosatovs, 2020-10-07:
@johanstokking @htdvisser please comment if you have opinions about this, otherwise, if you agree - let's remove the discussion label

johanstokking, 2020-10-07:
We do this guessing in V2 but it is quite limited with overlapping (join) frequencies across frequency plans. For example, AS920-923 vs AS923-925, while the device is only certified for one region.

It could be useful with US915 devices. With 8 channel gateways, basically the gateway's frequency plan drives which sub-band is being used. That could make sense indeed.

htdvisser, 2020-10-08:
I'm against guessing the band, but we can select a frequency plan _if the device owner set multiple "candidate" frequency plans in the device registration_.

I'm however not sure if we should really do this. Sure, it may be convenient for lazy device owners that don't feel like setting the field for the frequency plan, but in most situations it's just confusing, because such a "guessing" feature could make unexpected and inconsistent choices if we just look at the uplink frequency of the join request. 

The most common confusion is about 923.2 and 923.4 already mentioned by @johanstokking. These could be Join frequencies for the AS923/AS920-923/AS923-925 frequency plans. V2 makes the "safe" choice, and always picks AS923, but that is typically not what users want, because it only has 2 channels instead of 7. A small improvement to the AS923/AS920-923/AS923-925 issue can be made if the GS forwards the frequency plan that the gateway uses, so that the Network Server can match that against the "candidate" frequency plans in the device registration.

923.2 and 923.4 could also be Join frequencies of AU915 FSB6, so detecting the band is definitely impossible here, and guessing wrong means that we're sending Join-Accepts on the wrong channel.

So what's left is basically only US915. But then what if we'll at some point have networks with 8-channel gateways and 16-channel gateways? By "guessing" we can only be certain about one FSB. Depending on chance (the frequency) we would either pick the FSB that's used by both the 8-channel and 16-channel gateways, or pick the FSB that's only used by the 16-channel gateways. In both of these cases the behavior may look unexpected and inconsistent to the device owner. Here again, things could be slightly improved if the GS forwards the frequency plan that the gateway uses.

Finally, if the device owner can list the "candidates" in order of preference (i.e. the 16-channel is preferred, then the second FSB, then the first FSB, and then the others) we can probably make things less random.

rvolosatovs, 2020-10-08:
Please see the OP:
> Allow several frequency plans defined per-device from different bands with no join frequency overlap (i.e. the join frequencies should uniquely identify the frequency plan given the frequency plan [...]
The ABP implementation can come later and would need more elaborate handling. Perhaps we should not do it at all, in any case we should try this with OTAA first.

The proposal is to only allow such combinations, in which the join frequencies uniquely identify the band and frequency plan. The user still must set the frequency plan(s)

Please comment on the proposal and not on the general feature.

nejraselimovic, 2021-04-19:
@htdvisser @johanstokking can you comment the proposal?

htdvisser, 2021-04-20:
Purely commenting on the proposal to

> Allow several frequency plans defined per-device from different bands with no join frequency overlap

I'm against it, because (1) I don't think there's a big need, because there aren't that many multi-band devices, and it's really not that much work for users to pick a frequency plan for their single-band devices; and (2) we can't make it "always work", so it will be confusing.

---

Looking at the more general feature of "guessing frequency plans", I agree with @johanstokking that this could be useful to determine the used FSB(s) in fixed-channel frequency plans (such as US915, AU915 and CN470) when we don't know what FSB(s) the gateway(s) listen on. But if that's the only use case, I'd much rather have the gateways inform the NS about the frequency plan they use than the NS guessing it.

johanstokking, 2021-04-20:
Since this issue is specifically about the band, let's close this.

If we go with selecting the frequency plan within a band, based on the gateway(s) covering the end devices, we need GS to inform NS about the frequency plan to use (if none set in NS). This would need its own issue if we want this.

closed 2021-04-20 by johanstokking