TheThingsNetwork/lorawan-stack #3333: Make network-initiated downlink interval configurable

rvolosatovs, 2020-10-07:
#### Summary

Make network-initiated downlink interval configurable

#### Why do we need this?

To allow for network-initiated downlink rate-limiting by the customer

#### What is already there? What do you see now?

Hardcoded value in NS (https://github.com/TheThingsNetwork/lorawan-stack/blob/11cf05ba98cf6df44f064658ea4393183b0e7126/pkg/networkserver/networkserver.go#L57)

#### What is missing? What do you want to see?

Make this value configurable

#### How do you propose to implement this?

Add this as field in device MAC settings

#### How do you propose to test this?

integration testing

#### Can you do this yourself and submit a Pull Request?

yes

johanstokking, 2020-10-29:
> Add this as field in device MAC settings

Can we do this with a contextual override for the tenant?

rvolosatovs, 2020-10-29:
> > Add this as field in device MAC settings
> 
> Can we do this with a contextual override for the tenant?

Yes, but the simplest approach IMO would be to add this as a field to MAC settings and then make MAC settings available as per-tenant override

johanstokking, 2020-10-29:
How does this work with changing this value, with an existing set of devices? Since it's not MAC state, it doesn't get reset, so it would have to be set one-by-one, on the entire fleet of devices for it to be effective?

rvolosatovs, 2020-10-29:
We also have NS-wide default MAC settings, so for a particular tenant those could be set and would apply to all tenant's devices unless more specific (per-device) MAC settings are set.

neoaggelos, 2021-08-18:
For future reference, when adding new MAC settings (or generally new fields to the `EndDevice` message, the following changes are needed for the Network Server:

- Update `FieldIsZero()` check for the new field, e.g. https://github.com/TheThingsNetwork/lorawan-stack/pull/4547/files#diff-9085d0dbbf2898df8f7f8b12759aaef850f4e3e9cc88b970815cdced1d4219acR192-R193. Make sure to update all parent messages as well.

- Update `nsEndDeviceReadFieldPaths` and `nsEndDeviceWriteFieldPaths`, e.g. https://github.com/TheThingsNetwork/lorawan-stack/pull/4547/files#diff-b9fc0a941bb90c27aa1e4aa74faaf04907c0193cd22131627b12e8aac572f290R121

Ideally, this should be made part of CI/proto generation at some point.