TheThingsNetwork/lorawan-stack #3334: Gateway Server should wait before scheduling consecutive downlinks on a singular gateway

rvolosatovs, 2020-10-07:
#### Summary

Gateway Server should wait before scheduling consecutive downlinks on a singular gateway

#### Why do we need this?

To avoid situations when GS schedules downlinks too fast, such that gateway fails to transmit due to already being busy.

#### What is already there? What do you see now?

Not sure, but seems that there's no rate limiting and at least in one case a series of class C downlinks cause the gateway to respond with `RADIO BUSY` error.

#### What is missing? What do you want to see?

Gateway Server should wait for a reasonable amount of time before scheduling the next downlink in the series for the same gateway

#### How do you propose to implement this?

Add a timeout, similar to one we have in NS (https://github.com/TheThingsNetwork/lorawan-stack/blob/11cf05ba98cf6df44f064658ea4393183b0e7126/pkg/networkserver/networkserver.go#L57)

#### How do you propose to test this?

integration testing

#### Can you do this yourself and submit a Pull Request?

I rather not.

johanstokking, 2020-10-08:
Hmm, this is what Gateway Server does already:

https://github.com/TheThingsNetwork/lorawan-stack/blob/2aafb1a34f11781cbbd7726e8b0c2b7a097cde7e/pkg/gatewayserver/scheduling/scheduler.go#L333-L341

It could be that `timeOffAir` is too low. It is at least `QueueDelay`:

https://github.com/TheThingsNetwork/lorawan-stack/blob/2aafb1a34f11781cbbd7726e8b0c2b7a097cde7e/pkg/gatewayserver/scheduling/scheduler.go#L33-L36

---

Please provide an event log so we can look at the timestamps.

KrishnaIyer, 2020-10-08:
I'm pretty sure the schedular already does this

rvolosatovs, 2020-10-08:
Please see the email from a customer regarding this. You are both in the conversation.

chrisbouchat, 2020-10-08:
The GS appears to be buffering downlink requests for the configurable delay period and sending them in batches to the gateway. When I set the GS time delay to 1s and my test script to send downlinks every 250ms I see an average of 4 downlink requests arriving at the same time at the gateway. The gateway does not appear to be buffering or scheduling the downlinks and rejects/discards any that can not be sent while the radio is busy (I tried Multitech and RAK gateways). I attached and application event log and gateway log from the same time period.

Is this the intended behavior for the Gateway delay? Or is it intended to buffer + meter the downlinks?

If there are multiple gateways close to a group of devices is there any scheduling coordination of downlinks across the gateways?

[gateway_log.txt](https://github.com/TheThingsNetwork/lorawan-stack/files/5351180/gateway_log.txt)
[downlink_test_events.txt](https://github.com/TheThingsNetwork/lorawan-stack/files/5351181/downlink_test_events.txt)

johanstokking, 2020-10-27:
@chrisbouchat which packet forwarder are you using exactly, i.e. on the MultiTech? I need the version that it reports when it starts.

The idea of buffering downlink messages in GS is indeed because older packet forwarders do not have this buffering capability. So if GS would send downlink messages immediately to the gateway, and there would be multiple, old downlink messages are being overwritten. Worst case, all downlink messages are being overwritten all the time, such that the gateway is not sending any downlink.

More modern packet forwarders do have a downlink message queue, called JIT queue.

johanstokking, 2020-10-29:
> @chrisbouchat  The gateway does not appear to be buffering or scheduling the downlinks and rejects/discards any that can not be sent while the radio is busy (I tried Multitech and RAK gateways)

I had a closer look at the logs. The gateway does in fact have a JIT queue, but apparently GS is scheduling too short:

  |   |   | Diff
-- | -- | -- | --
3458319307 | SF12BW500 | 14 |  
3459479163 | SF12BW500 | 14 | 1159856
3459565993 | SF12BW500 | 14 | 86830
3460143163 | SF12BW500 | 14 | 577170
3460579650 | SF12BW500 | 14 | 436487
3461162905 | SF12BW500 | 14 | 583255
3461759320 | SF12BW500 | 14 | 596415
3462025385 | SF12BW500 | 14 | 266065
3462754238 | SF12BW500 | 14 | 728853
3463563844 | SF12BW500 | 14 | 809606
3463690803 | SF12BW500 | 14 | 126959
3464204649 | SF12BW500 | 14 | 513846
3464800435 | SF12BW500 | 14 | 595786
3466191424 | SF12BW500 | 14 | 1390989
3466308537 | SF12BW500 | 14 | 117113
3466590702 | SF12BW500 | 14 | 282165
3467185630 | SF12BW500 | 14 | 594928
3467865726 | SF12BW500 | 14 | 680096
3468370568 | SF12BW500 | 14 | 504842
3468701822 | SF12BW500 | 14 | 331254
3469089234 | SF12BW500 | 14 | 387412
3469744405 | SF12BW500 | 14 | 655171
3470217526 | SF12BW500 | 14 | 473121
3470852184 | SF12BW500 | 14 | 634658
3471663309 | SF12BW500 | 14 | 811125
3471759194 | SF12BW500 | 14 | 95885
3472253208 | SF12BW500 | 14 | 494014
3473065700 | SF12BW500 | 14 | 812492
3473562766 | SF12BW500 | 14 | 497066
3474163029 | SF12BW500 | 14 | 600263
3475072789 | SF12BW500 | 14 | 909760
3475283406 | SF12BW500 | 14 | 210617
3476082378 | SF12BW500 | 14 | 798972
3476369691 | SF12BW500 | 14 | 287313
3477059761 | SF12BW500 | 14 | 690070
3477671610 | SF12BW500 | 14 | 611849
3478173977 | SF12BW500 | 14 | 502367
3478689042 | SF12BW500 | 14 | 515065
3478987854 | SF12BW500 | 14 | 298812
3479700397 | SF12BW500 | 14 | 712543
3480167916 | SF12BW500 | 14 | 467519
3480977021 | SF12BW500 | 14 | 809105
3481160052 | SF12BW500 | 14 | 183031
3481850955 | SF12BW500 | 14 | 690903
3482758837 | SF12BW500 | 14 | 907882
3482864103 | SF12BW500 | 14 | 105266
3483465844 | SF12BW500 | 14 | 601741
3483656616 | SF12BW500 | 14 | 190772
3484864035 | SF12BW500 | 14 | 1207419
3485187166 | SF12BW500 | 14 | 323131
3485598054 | SF12BW500 | 14 | 410888

You can clearly see the `REJECTED` errors when the diff is low. So yes, this is a bug in GS and might be related to the time-on-air calculation.

johanstokking, 2021-02-10:
@chrisbouchat are these issues (only) with class C downlink scheduling without absolute time?

johanstokking, 2021-02-12:
I spent half the day looking into this. There's at least one fix, although the chance is small that that is the root cause.

---

### Attempt 1

DevAddr: 260837AE
Class C downlink with no absolute time

Uplink token:

```json
{
  "ids": {
    "gateway_ids": {
      "gateway_id": "00-9181003t19c00586",
      "eui": "60C5A8FFFE7662B2"
    }
  },
  "timestamp": 3398136108,
  "server_time": "2020-10-08T20:09:06.339638892Z",
  "concentrator_time": "162311926060000"
}
```

Network server time: 2020-10-08T20:10:06.684265770Z 
Schedule delay: 0.999980090s
Timestamp: 3459479163
Time-on-air: 288768000

Starts: 162311926060000 - 3398136108000 + 3459479163000 = 162373269115000
Ends: 162373269115000 + 288768000 = 162373557883000

### Attempt 2

DevAddr: 2608152C
Class C downlink with no absolute time

Uplink token:

```json
{
  "ids": {
    "gateway_ids": {
      "gateway_id": "00-9181003t19c00586",
      "eui": "60C5A8FFFE7662B2"
    }
  },
  "timestamp": 3401231164,
  "server_time": "2020-10-08T20:09:09.436794776Z",
  "concentrator_time": "162315021116000"
}
```

Network server time: 2020-10-08T20:10:06.770839155Z
Schedule delay: 0.999986804s
Timestamp: 3459565993
Time-on-air: 288768000

Starts: 162315021116000 - 3401231164000 + 3459565993000 = 162373355945000
Ends: 162373355945000 + 288768000 = 162373644713000

### Observations

#### Uplink tokens

Concentrator time delta: 3095056000 (compare `concentrator_time`)
Server time delta:       3097155884 (compare `server_time`)

That's acceptable due to network delay.

#### Schedule

Timestamp delta: 86830 µs
Server time delta: 86573.385 µs

So: the scheduling diff is the delta, meaning that the previous emission is not being taken into account.

Start of attempt 2 should be at 162373557883000 + 288768000 + 30000000 = 162373876651000
> previous ends + time-on-air + queue delay of 30 ms

Time between schedules according to Network Server events is 86573.385 µs. The trace region `schedule transmission` (`ScheduleAt`) is under 100 µs and the trace region `schedule transmission any time` (`ScheduleAnytime`) is under 100 µs too in The Things Stack Cloud `nam1`. So a race is unlikely, but cannot be entirely ruled out.

There could be a broader issue, and that there's a different `Scheduler` instance being used. This can also be the cause for https://github.com/TheThingsNetwork/lorawan-stack/issues/3487. We're waiting for additional logs there that correlate the `Scheduler` instance for each scheduling attempt.

johanstokking, 2021-02-12:
Looping in @adriansmares for another pair of eyes.

The problem is that there are two consecutive downlink messages scheduled that are clearly overlapping. The second attempt does not see the first attempt, and therefore creates a scheduling conflict.

There is a race condition which is fixed in #3794 but the chance that that happens is very small. It cannot be ruled out though.

Any other thoughts by `emissions` in `Scheduler` would not be in sync?

KrishnaIyer, 2021-02-15:
> There could be a broader issue, and that there's a different Scheduler instance being used. 

Hmm that seems unlikely. A gateway can connect to only one GS instance (due to ID claiming). And one connection only creates one scheduler instance ( and hence one slice of emissions).

johanstokking, 2021-02-17:
> Any other thoughts by `emissions` in `Scheduler` would not be in sync?

This turned out to be the case. This is now closed by #3794.

> but the chance that that happens is very small

The chance wasn't small because multiple goroutines were waiting to acquire the read lock, while they were actually writing. As soon as something else released the write lock, those writing goroutines were exactly in sync, corrupting `emissions`.

closed 2021-02-17 by johanstokking