TheThingsNetwork/lorawan-stack #3342: Strategy for handling backwards-incompatible optimizations in registries

rvolosatovs, 2020-10-09:
#### Summary

We need to discuss strategy for handling backwards-incompatible optimizations in registries

#### Why do we need this?

Sometimes we can do optimizations in registries, which depend on a particular storage scheme of the data. E.g. #2837 

#### What is already there? What do you see now?

In #3331 `CompatibilityVersion` was introduced as part of NS Redis registry config. I did not add a way to set this value via config for now, since that would mean we'd have to stay compatible with this forever and I would like to discuss first.

#### What is missing? What do you want to see?

A way to set minimum version registry should be compatible with. This can be the stack version, this could also be registry-specific version, there are different options.

#### How do you propose to implement this?

I think the simplest option from user perspective is to have one shared `compatibility-version` in top-level config, this value is then propagated to where it's needed. This version is stack version, namely, the version of the stack when it was first ever run.
So e.g. if Bob deploys 3.2.0, he sets `compatibility-version` to 3.2.0 and can be sure that everything will work(perhaps, not optimally, but it will) even if he then updates the stack to 3.42.0.
Now, if Bob wants to benefit from hypothetical crazy optimizations in version 3.42.0, he would call `ttn-lw-stack migrate` (this is not a suggestion on command naming/API, please treat this as just an example), which would migrate the structure of the registr{y,ies} and then Bob would set `compatibility-version` to 3.42.0 and rest assured that registry is performing in the most optimal way possible in version 3.42.0. Note, that even if `compatibility-version` was not updated after migration, registry should still stay compatible by virtue of backwards-compatibility commitment. The `compatibility-version` is merely an indication of minimum version registry should be compatible with, i.e. it's an indication of which optimizations can be performed and which cannot.

Other options:
- Store the version in registry at a separate key and perform migrations on-the-fly - IMO this is quite error-prone and will interfere with ability to downgrade in required, so :-1: 
- Have a version per-registry or guard the various version-dependent optimizations/features by individual config options - possible to do, but the configuration can become cumbersome with time, the easiest solution then would be to introduce a config option tied to stack version, which customers could set and derive these individual versions/options from that. IMO this is just a more complex way to do proposed implementation with zero added value - registries can perfectly derive what's necessary from the stack version. :-1: 

#### How do you propose to test this?

Try upgrading the stack from different versions with different `compatibility-version` values

#### Can you do this yourself and submit a Pull Request?

yes

johanstokking, 2020-10-10:
> @rvolosatovs  since that would mean we'd have to stay compatible with this forever and I would like to discuss first

It is forever in V3:

> 2. We will upgrade storage from older versions within the major version via migrations. This means that you can migrate an older setup without losing data.

The compatibility commitment is holy: `3.42.0` and `3.468924396.0` are all "within the major version".

rvolosatovs, 2020-10-12:
> > @rvolosatovs  since that would mean we'd have to stay compatible with this forever and I would like to discuss first
> 
> It is forever in V3:
> 
> > 1. We will upgrade storage from older versions within the major version via migrations. This means that you can migrate an older setup without losing data.
> 
> The compatibility commitment is holy: `3.42.0` and `3.468924396.0` are all "within the major version".

How is this related to this issue? Sure, this only concerns v3, since in v4 we can do backwards-incompatible changes.

johanstokking, 2020-10-12:
It is related because we don't make a claim of backwards compatibility. In minor versions there may be manual database migrations (see point 3 of the compatibility commitment), but it is not stated that they can be undone. In fact, you can require a database migration as part of upgrading to a new minor; you don't have to stay backwards compatible. So, I would not spend too much time on backward compatibility on the storage level; forward compatibility is guaranteed without manual steps in patches and potentially with manual steps in minors.

johanstokking, 2020-10-12:
Pushing to November and probably Backlog because I don't see the urgency for this monht.

rvolosatovs, 2020-10-12:
> Pushing to November and probably Backlog because I don't see the urgency for this month.

FTR, until we have this, new deployments (and existing ones) cannot benefit from some optimizations in #2837

johanstokking, 2020-10-12:
I don't think we need "this", i.e. this issue: maintain backwards compatibility of older storage schemes. I do think we need a migration tool, just like we have `is-db migrate`.

htdvisser, 2020-10-13:
So, just as we usually have a `ttn-lw-stack is-db migrate` that needs to be executed before minor version upgrades, we tell operators to run `ttn-lw-stack ns-db migrate` before/when/after deploying a new minor version. Migrations only need to be "forward", because our compatibility commitment doesn't say that you can downgrade to a previous minor.

For reference: the way migration works in v2 is that we store a "version" field in each object in Redis, and we migrate "on the fly" when needed. We also had a `migrate` command that would loop over all objects, and migrate every object that wasn't already migrated "on the fly".

From what I understand, @rvolosatovs's suggestion is to introduce the concept of a "minimum compatibility version" that indicates the minimum version of any object in the registry. If this value is above some version, the Network Server can disable some fallbacks (trying the old format). 

I think it's a good idea, and actually, I would even go further. Don't make it config, hardcode it. A `v3.10` Network Server can work with a `v3.9+` database through fallbacks, but any data before `v3.9` needs to be migrated. This way, operators can deploy the `n+1` minor without disrupting operations, and then run the migration tool to upgrade all objects in the registry to `v3.10`. When the migration tool is done, it SETs `v3.n+1` as the "minimum version" in Redis. When a `v3.n` Network Server starts, it checks this version. If it's not `n` or `n-1` it throws a clear error: migrate your database. If it's `n`, it can disable the fallbacks.

Does this sound reasonable? Or does it only make sense in my head?

johanstokking, 2020-10-13:
It makes sense but then only with the purpose of reducing downtime.

htdvisser, 2020-10-13:
Yes, and rolling back if the deployment fails (assuming that you didn't skip a minor)

johanstokking, 2020-11-04:
OK, let's go with:

> Don't make it config, hardcode it. A `v3.10` Network Server can work with a `v3.9+` database through fallbacks, but any data before `v3.9` needs to be migrated. This way, operators can deploy the [...]

@rvolosatovs please implement this when it becomes necessary.

closed 2020-11-04 by johanstokking