TheThingsNetwork/lorawan-stack #3360: Retry failed downlinks on Network Server level

rvolosatovs, 2020-10-14:
#### Summary

Retry failed downlinks on Network Server level

#### Why do we need this?

To simplify life of application developers

#### What is already there? What do you see now?

Network Server retries if e.g. class B/C absolute time is valid, but gateway does not have sync yet by looking at the error received immediately from the GS on scheduling

#### What is missing? What do you want to see?

Retry downlinks, which were scheduled successfully, but failed to actually be transmitted by the gateway

#### How do you propose to implement this?

Add a downlink task on receiving a "downlink failed" TX event after correlating it to the downlink and, hence, the device

#### How do you propose to test this?

Not sure how exactly yet, but with physical setup + unit tests

#### Can you do this yourself and submit a Pull Request?

yes

rvolosatovs, 2020-10-14:
Blocked by #76

nejraselimovic, 2021-06-07:
@adriansmares #76 is merged, so I guess this is not blocked anymore. Can you tell me how is this issue different from #3642? Is it because this one is not for confirmed downlinks only?