TheThingsNetwork/lorawan-stack #3394: Poorly formatted password errors

benolayinka, 2020-10-22:
#### Summary

Error text for password errors on account creation is missing some words

#### Steps to Reproduce

1. Go to the create account page in the console
2. Enter passwords that don't match or are too short

#### What do you see now?

[image]

#### What do you want to see instead?

"Password must be at least 12 characters", or something that is a complete sentence.

#### Environment

3.10

#### Can you do this yourself and submit a Pull Request?

Let me know if you need help with the text @kschiffer

kschiffer, 2020-10-22:
Indeed. These errors come from the backend, so I guess since we don't want to change the message there, we'd need to define our own frontend messages for these (using the message code `error:pkg/identityserver:password_strength_min_length`), which then conform to the wording guidelines we have.
cc @htdvisser

htdvisser, 2020-10-22:
@johanstokking probably knows more, this since he implemented it, but I guess those errors should have been wrapped in a "password not secure enough" error.